lgsvl/simulator #762: About Cyberlog Parsing

yumianhuli2, 2020-05-15:
Is there a cyberlog parsing feature or script in this project？Thank you！

EricBoiseLGSVL, 2020-05-15:
@yumianhuli2 Is [this](https://github.com/ApolloAuto/apollo/issues/8388) what you are asking about?

yumianhuli2, 2020-05-15:
> @yumianhuli2 Is [this](https://github.com/ApolloAuto/apollo/issues/8388) what you are asking about?

Yes similar to ros record for playback log, does cyberlog have such a playback tool in this proj?Thank you!

lemketron, 2020-05-15:
> does cyberlog have such a playback tool in this proj?Thank you!

@yumianhuli2 Yes, Apollo has [Cyber RT Developer Tools](https://github.com/ApolloAuto/apollo/blob/master/docs/cyber/CyberRT_Developer_Tools.md) including cyber_recorder:

> cyber_recorder is a record/playback tool provided by Apollo Cyber RT. It provides many useful functions, including recording a record file, playing back a record file, splitting a record file, checking the information of record file and etc.

yumianhuli2, 2020-05-16:
> @yumianhuli2 Yes, Apollo has [Cyber RT Developer Tools](https://github.com/ApolloAuto/apollo/blob/master/docs/cyber/CyberRT_Developer_Tools.md) including cyber_recorder:

Hi! how can I use it in this simulator in unity?

EricBoiseLGSVL, 2020-05-19:
@yumianhuli2 It's not used in Simulator but in Cyber

closed 2020-05-19 by yumianhuli2