lgsvl/simulator #764: New starter with LGSVL have some questions

SenRamakri, 2020-05-15:
Hello All,

I'm new to LGSVL and have been trying to run Autoware.AI(1.13 version) with LGSVL. Im running into few issues and thought of asking them here, any help would be appreciated.

1. Can I run LGSVL without GPU and still expect things to work although I understand that things would be terribly slow? I'm currently trying to run it on a 12 core Ubuntu 18.04 system without GPU.
By the way, Im not trying to do anything fancy at this point. Im initially starting to only see if I can communicate with the Websocket/API?
2. Currently I downloaded the Lgsvl release version(2020.01), and just executed the "simulator" executable. This brings up the config window and after configuring it then opens the window with "Open Browser" button. And in the browser(with URL http://localhost:8080/#/), in Simulations tab I clicked on the edit button for "BorregasAve(with Autoware)" and ensured that the bridge string is set to "localhost:9090". Then I selected "BorregasAve(with Autoware)" and clicked on the play button(in the bottom banner). This obviously started the simulation and I could see the LGSVL window showing the car animation. At this point, I expected a server/simulation to be listening in my system at port 9090. But if I do netstat/nmap I don't see anything listening on localhost:9090 (or even the API port which is 8181) although I do see port 8080 listening by simulator process. My intention was to start using Websocket/JSON APIs at URL http://localhost:9090/simulator/launch at this point and obviously the connection fails as there is no port listening. May be my understanding is wrong on when/how to configure Websocket to start listening. And looking at Player.log I see entries like "|Fatal|WebSocket.connect|An address incompatible with the requested protocol was used", so Im thinking this indicates something wrong with init-ing websocket. Attached is the full log. Any help to resolve this websocket issues would be great.

Thanks in advance.

[Player.log](https://github.com/lgsvl/simulator/files/4636866/Player.log)

SenRamakri, 2020-05-15:
As a sanity check, I changed the bridge string to "127.0.0.1:9090" and the error string in Player.log changed to "|Fatal|WebSocket.connect|Connection refused"

SenRamakri, 2020-05-15:
Im looking into LGSVL code and I guess that the Websocket is actually a client, is that correct? So should I be starting the websocket rosbridge for this Websocket APIs to work?

lemketron, 2020-05-15:
> Im looking into LGSVL code and I guess that the Websocket is actually a client, is that correct? So should I be starting the websocket rosbridge for this Websocket APIs to work?

That is correct. According to the [Running with Autoware.AI](https://www.lgsvlsimulator.com/docs/autoware-instructions/) documentation for LGSVL Simulator:

> Autoware communicates with the LGSVL simulator using the rosbridge_suite, which provides JSON interfacing with ROS publishers/subscribers. The official Autoware Docker containers have rosbridge_suite included.

After installing Autoware.AI and ROS, and launching the Autoware Runtime Manager, you will eventually click "Sensing" which also launches rosbridge.

Please refer to the LGSVL/Autoware instructions for full details: https://www.lgsvlsimulator.com/docs/autoware-instructions/

SenRamakri, 2020-05-16:
Thanks for the response.@lemketron. Yes, I was able to run rosbridge_server (rosbridge_websocket) and things are working now. I no longer see websocket connection error in Player.log. As of now, below is what I have.
[LGSVL Simulator]<==working==>[ROS Bridge]<==working==>[Autoware]
And when look at Player.log it shows LGSVL(Client 0) connected and subscribed to /vehicle_cmd topic.
With this working, I have a test script in Python which essentially connects to http://localhost:9090/simulator/launch and sends the configuration Json at https://www.lgsvlsimulator.com/docs/autoware-json-example/
I was expecting a valid reply from LGSVL, but what I got was a 404 error. Or will this JSON config be used in a different contenxt? Anyway thanks for your pointers @lemketron

lemketron, 2020-05-16:
Please check out the [Quickstart](https://www.lgsvlsimulator.com/docs/python-api/#quickstart) instructions for using the Python API which should help explain the startup process. More specifically check out steps 3-7 to see if you are able to run the example script as documented.

SenRamakri, 2020-05-17:
@lemketron - Thanks again for the docs pointer. That seems to have great info I have been looking for. Is there any docs on Json spec/API as well?

lemketron, 2020-05-18:
@SenRamakri Are you looking for docs on the sensor configuration JSON? There are several sample sensor configurations included in our documentation for various AD stacks including Autoware.AI, Autoware.auto, Apollo 5.0, and Apollo 3.0.

SenRamakri, 2020-05-18:
@lemketron  - Yes, the sensor configuration API as well as the API for talking to API-Port(8181).
I did notice the sample Json configuration provided(for Autoware.AI) but I was looking for documentation/specification on the API, if its available.

martins-mozeiko, 2020-05-18:
I think there is misunderstand what each ports does and what kind of information is communicated.

Simulator listens on port 8080 for requests from web interface. All the actions you do in web interface results in GET/POST/etc.. calls to http server on port 8080. Protocol is internal and not documented. Although you can easily see what kind of data gets sent either from C# and JavaScript source code, or by simply looking in Network traffic in your browser.

Simulator lists on port 8181 for Python API commands. This API is documented on Python level here: https://www.lgsvlsimulator.com/docs/python-api/ It includes all capabilities that this endpoint supports. Protocol itself is done by sending JSON and is internal and not documented. But again - you can easily examine it by looking in C# and Python source code.

Port 9090 is used for communicating to bridge (ROS or Cyber). Simulator here is only client that connects to bridge and uses bridge specific protocol (json or binary) to send and receive ROS/Cyber messages. You do not need to send or receive anything on this port. All the interaction to this should happen with ROS commands (rostopic pub/echo) or Cyber commands or their API's.

Sensor JSON from Autoware/Apollo pages is just a config you need to put in Vehicle configuration here: https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle It is not related to any of API endpoints.

SenRamakri, 2020-05-18:
@martins-mozeiko - This is really great response and clarified most of my questions. And thanks for the Urls provided, they look to be providing the info Im looking for and I'll be reviewing them in detail.
It appears you have answered most of my current questions and I'll close this issue now thankfully :-).

closed 2020-05-18 by SenRamakri

SenRamakri, 2020-05-18:
@martins-mozeiko - Just a thought, if you could somehow capture whatever you mentioned above in your comment in the documentation as well, that would be really helpful.

rongguodong, 2020-05-18:
> @martins-mozeiko - Just a thought, if you could somehow capture whatever you mentioned above in your comment in the documentation as well, that would be really helpful for future users.

That's a great idea. This information is a good topic under "Advanced" Section at the bottom of the document page. :)