lgsvl/simulator #765: How export map to AssetBundle?

miluit01, 2020-05-18:
Hey guys,

I want to change the traffic light positions in Shalun map and export the scene in the AssetsBundle. After export the environment_shalun file is created but can not be use in the simulator. 

What I did:

git clone https://github.com/lgsvl/Shalun.git in Assets/External/Environment

build the Shalun via Simulator->Build->select Shalun in environment tab->click build

After few minutes the environment_shalun is created in AssetsBundle folder

Open the simulator add the new map, give the absolute path of environment_shalun file and the new map is added successfully.

But: I setup a new simulation which is using the new environment very soon after start:

[image]

Many thanks for any help!

EricBoiseLGSVL, 2020-05-18:
@miluit01 Are you loading the new asset with Editor or a build?  If a build, what version is it and what release version are you using from GitHub?

miluit01, 2020-05-19:
> @miluit01 Are you loading the new asset with Editor or a build? If a build, what version is it and what release version are you using from GitHub?

Hi Eric,

I am using a release Editor 2020.03 and build the Shalun in a download source of master branch.
Should I build the Editor of the release 2020.03 tag source code and build Shalun with this release tag as well?

Many thanks for your reply!

EricBoiseLGSVL, 2020-05-19:
@miluit01 Yes, use the release branch, not master

miluit01, 2020-05-19:
> @miluit01 Yes, use the release branch, not master

Hi Eric,

I switched to release-2020.03 branch and build AssetsBundle and simulator as well. But it still has the same issue. I checked the output windows in Editor and get the following error:

[image]

I tried start the simulator in Editor via LoadScene and the built binary directly. Both have the same problem.

miluit01, 2020-05-19:
> @miluit01 Yes, use the release branch, not master

I checked Loader.cs line 549 and get the feeling that somehow the manifest of AssetsBundle doesn't match the BundleConfig.MapBundleFormatVersion = 2

`                                if (manifest.bundleFormat != BundleConfig.MapBundleFormatVersion)
                                {
                                    zip.Close();

                                    // TODO: proper exception
                                    throw new ZipException("BundleFormat version mismatch");
                                }`

I'm wondering why this number is wrong, if I am using the right version of branch and asset. Is it possible to check which manifest version has Build took to put in the AssetsBundle?

miluit01, 2020-05-19:
Hi Eric,

I found out, that the built environemt_shalun file is not correct generated. See previouly error message in Editor.
To compare I wget https://assets.lgsvlsimulator.com/5509eb186899e1f1596bd131e9614303e95aa19e/environment_SingleLaneRoad
The downloaded file could be errorless opened with Archiver Manager.
But the built one couldn't.
I think it was something going wrong during build process.

EricBoiseLGSVL, 2020-05-19:
@miluit01 You need to have both linux AND windows build support downloaded from Unity.  Bundles are required to be cross platform

miluit01, 2020-05-20:
@EricBoiseLGSVL Hi Eric, After add "Window build support" AssetsBundle could be generated and used in simulator. Many thanks for your help!

closed 2020-05-20 by EricBoiseLGSVL