lgsvl/simulator #766: How to annotate map Shalun

miluit01, 2020-05-18:
Hey guys,

I am following the doc try to annotate Shalun map.

When I am opening Assets/External/Environment/Shalun/Pretabs/MapShalun and just see a map without road only the traffic lights and signs show up in the gui.

[image]

Many thanks for any help!

EricBoiseLGSVL, 2020-05-18:
@miluit01 You are in Prefab view in Editor.  Go back to scene view to edit annotation.  Also be sure to enable gizmos in Editor to see the annotation.

[image]

miluit01, 2020-05-19:
Hi Eric,

perfect. It works! You save my day!

p.s.: some how my git lfs worked not correctly. I re-git lfs clone everything. Then activate Gizmos and annotation is show up.

Many thanks!

closed 2020-05-19 by miluit01