lgsvl/simulator #768: Data gathering -- how to increase frequency of the data received from the LGSVL

dr563105, 2020-05-18:
Hello!

I don't if this is LGSVL or ROS problem. I'd like to try to seek help here. 

In my data collection node, I use [approx time sync](https://github.com/ros2/message_filters/blob/master/src/message_filters/__init__.py#L242) class from the [ros2_message_filters](https://github.com/ros2/message_filters). In its argument I adjusted the queue size to 70 (as mentioned here in step [1.4](http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers)).

`ts = message_filters.ApproximateTimeSynchronizer([sub_center_camera, sub_left_camera, sub_right_camera, sub_control], 70, 0.03)
ts.registerCallback(self.callback)`

This change made a big difference making the data appear in the terminal a second or two after pressing `play` in the simulator. I also see that topic backs up in my latest trials. Data get saved for 5-10 more frames after stopping the simulator. Is there a way to manipulate this argument in a better way? As far as I looked, I couldn't find buffer size argument. 

However, the number of entries I expect per minute is smaller. My target is 100,000 data entries. The standard camera nodes have a frequency of 15Hz set in the JSON sensor parameter(getting CompressedImages) . And the control topic has steering, acceleration and few other state info. Clicking the `plug` icon in the Simulator, gives me a constantly changing frequencies of both camera and the topic. Sometime the number raises to 1200Hz. What do these numbers indict? 
 
Right now I get about 5-6 entries per second(5Hz). With that in calculation I would need to drive for more than 5 hours. That seems like a daunting task. Is that how it works in general?

So is there a way to frequency of receiving the data from a topic through rosbridge and storing rate faster? 
Should I increase the websocket buffer to a larger number in [Rosbridge.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosBridge.cs#L48)?

Thank you.

EricBoiseLGSVL, 2020-05-18:
@dr563105 we will need to look into this.

dr563105, 2020-05-18:
Thanks Eric. You can use the latest Lanefollowing tutorial, Collect script. Just adjust the value of queue size and slop as I did. Though I expanded the control topic to include more info, it shouldn't be a problem to verify.

dr563105, 2020-05-19:
One thing I did was increase the queue size to 100 and slop to 2.5 seconds. In the CSV file, about 10 entries per second. So it reduces the time to my data target to 2 hrs 35 mins driving time.

rongguodong, 2020-05-20:
There are some factors limited the cap of rate you can achieve: Each camera sensor has a `Frequency` which is set to 15 by default; But even if you raise it, it also depends on how many frames your machine can render for three cameras. Furthermore, even if your machine is powerful enough to generate many frames per second, it is still limited by how much data you can send out via bridge.

We currently have a bandwidth limit of the ROS bridge. I believe 10Hz is already the max rate you can get for three cameras. We are looking into workaround to send out camera images without bridge. If we have a better way to do that, it may be included in the future release of the simulator.

With that, if the sending rate is 10Hz, no matter what you change on the receiving side, the rate will be capped at 10Hz.

dr563105, 2020-05-20:
Thank you @rongguodong.  

Do you mean to say since I use three camera sensors, it is slowing down my machine capability to process faster? What performance impact would I have if I add LIDAR into the mix? 

I tried to change the buffer size of the web socket in Rosbridge.cs. It didn’t seem to have helped. 

> We are looking into workaround to send out camera images without bridge.

Is this something I can accomplish in a short time say a week or two? Just a small addition to help me make progress in the short term? 
Where can I look?

rongguodong, 2020-05-20:
@dr563105 If you add a Lidar, it definitely affect the performance. However, if your machine is powerful enough, most probably the bottleneck is the bridge bandwidth. So the publishing rate may not be very different with and without Lidar.

BTW: We have implemented the work around for Lidar sensor. The newly introduced [VelodyneLidarSensor plug-in](https://www.lgsvlsimulator.com/docs/lidar-plugin/) uses UDP to directly send Lidar data to the server (where AV stack is running), and does not use the bridge at all. You may try VelodyneLidarSensor plub-in, or build your own Lidar Sensor plug-in if you are using other Lidar models.

We are looking into similar workaround for camera sensor as well. If you come with a good solution, you are welcome to create a pull request to contribute to our simulator.

dr563105, 2020-05-20:
Incidentally I even increased the UDP limit of Ubuntu to a higher number 5228220(not sure on the exact number). I already use the Lanefollowingsensor plugin. I will look into VelodyneLidarSensor plug-in. 

Thanks again. This is really encouraging that I can reduce the data gathering time.

dr563105, 2020-05-26:
Closing this issue for now as I couldn’t solve it before my data collection task. I reckon I had to drive around for more than four hours.

closed 2020-05-26 by dr563105

EricBoiseLGSVL, 2020-05-26:
@dr563105 sorry we could not resolve this