lgsvl/simulator #772: Improving measures of lidar simulation

bxdd, 2020-05-21:
When I run the lgsvl lidar, I found that this simulator has two shortcomings and one error that needs to be corrected. If this error is corrected, then both of the simulator's shortcomings can be resolved.

I am an independent developer of lidar for the carla simulator. I rewrote the lowest-level code of carla to realize the GPU acceleration of lidar in carla. I found that an error in lgsvl is consistent with the situation I encountered during development.

## Let me talk about the error first:

 I believe that the algorithm of lgsvl intersecting the laser and depth image is problematic, which makes the point cloud effect become a polygon instead of a circle. Note that when finding intersections, we cannot directly use Euler angles to get the coordinates, but need real intersections. In fact, the intersection of a laser and demth image is a curve. But in lgsvl, the intersection of laser and image is a straight line.

The following picture is the correct track of laser on the image：
[image]

The following figure is the polygon effect of lgsvl point cloud：
[image]

## Let me talk about the shortcomings：
Firstly, I found that the lgsvl radar is obtained by taking turns rendering through 24 depth cameras. Similar to no vertical synchronization, this makes the simulator point cloud tear. If the previous error is resolved, only 3 depth cameras are needed. This does not make the rendered image triangular, but a perfect circle.
Point cloud picture tear：
[image]

Secondly, I found that The lidar simulation delay of lgsvl is very large. This is because the rendering thread lags behind the main thread. However, if we solve the previous error, we only need to use 3 depth cameras to replace the original 24 cameras. This will greatly reduce the simulation delay.

EricBoiseLGSVL, 2020-05-21:
@bxdd Thanks for the help.  I'm going to ping @rongguodong who has been improving our sensors.

rongguodong, 2020-05-21:
@bxdd Which version are you using? The screenshot seems to be from an old version. This should have been fixed in newer versions. Can you try 2020.03?

rongguodong, 2020-05-21:
Using 3 cameras will greatly increase the texture size and thus make the GPU-readback slower. With more cameras, we can have more smaller texture read-backs running in parallel.

bxdd, 2020-05-22:
> @bxdd Which version are you using? The screenshot seems to be from an old version. This should have been fixed in newer versions. Can you try 2020.03?

Thanks for your answer, I'll try the newest version.

bxdd, 2020-05-22:
> > @bxdd Which version are you using? The screenshot seems to be from an old version. This should have been fixed in newer versions. Can you try 2020.03?
> 
> Thanks for your answer, I'll try the newest version.

> Using 3 cameras will greatly increase the texture size and thus make the GPU-readback slower. With more cameras, we can have more smaller texture read-backs running in parallel.

But will too many depth cameras cause point cloud tears? One suggestion is to develop two modes. One mode enables vertical synchronization to focus on point cloud quality, and the other mode focuses on frame rate.

rongguodong, 2020-05-22:
What do you mean by "point cloud tears"?

rongguodong, 2020-06-01:
> > @bxdd Which version are you using? The screenshot seems to be from an old version. This should have been fixed in newer versions. Can you try 2020.03?
> 
> Thanks for your answer, I'll try the newest version.

Does 2020.03 works well? If so, we can close this issue.

closed 2020-06-09 by EricBoiseLGSVL