lgsvl/simulator #775: Creating a new environment/map based real location

eyadshak, 2020-05-27:
Hello LG Team,

I have been trying to figure out how to create a new map for use in LGSVL, I've seen the tutorials and some posts regarding this and I need some clarifications.

Its possible to import a point cloud file according to the tutorial (https://www.youtube.com/watch?v=6r7K3-dV1iM), but the end result is very blurry and would not work. Is it possible to generate a better looking environment if I use better point cloud data? And what type/requirements of point cloud data would be needed to do that? 

According to #251 to generate a map like SanFrancisco, its possible to use a 2d map and then build the road, buildings, traffic light, etc.

What would be the best way to recreate a real life location (a university campus)?

Thank you!

EricBoiseLGSVL, 2020-05-27:
@eyadshak Hello, and welcome to LGSVL.  There are some ways to improve point cloud data but it is currently in development.  As for SF, that was built with the help of 3D artists and tools.  It really depends on the fidelity that you need for your use case.  Are you looking for hyper realism, accuracy or both?
@daviduhm @lemketron

eyadshak, 2020-05-27:
Thanks for the reply Eric, we are looking for something that is similar to the maps currently available with the LGSVL simulator, so I am guessing both. 

What would be the correct approach, we were exploring the idea of capturing a 3D map of the campus by aerial drones and maybe importing that into unity, would that be possible, would you know what kind of data would be needed?

Thanks for your help

EricBoiseLGSVL, 2020-05-28:
@eyadshak For point cloud to mesh you will need software to generate meshes from point could data and images.  Then you will need to cleanup mesh and apply textures from images.  Another way is to have aerial images from drones or google maps and place on a plane in Unity.  Then create/place buildings/decoration.  This is faster and easier but not as accurate.  Lastly, you can clean up the point cloud data and use our point could features in simulator.

eyadshak, 2020-05-31:
Thank you for the explanation, made things a lot clearer. If anyone has any advice on any additional tools to use to ease things for me I would greatly appreciate it.

closed 2020-05-31 by eyadshak

EricBoiseLGSVL, 2020-06-01:
@eyadshak Glad to help.  @daviduhm and @lemketron might have tools you could look at