lgsvl/simulator #776: traffic light control

gyp199009, 2020-05-29:
Hello, 
I wander is there any way to control the traffic light in realtime (including getting & setting the signal), just like controlling the EGO vehicle through rosbridge.
Thanks in advance!

ntutangyun, 2020-05-29:
Hi @gyp199009 , you could check out the docs [here](https://www.lgsvlsimulator.com/docs/python-api/#controllable-objects).

gyp199009, 2020-06-01:
Thank you for your information, but what I want to implement is controling the traffic light in realtime (during the simulation), is it possible to achieve that by using Python API ?

ntutangyun, 2020-06-01:
@gyp199009 I've never tried, but maybe you could try the following:

```
// set to green in real-time
control_policy = "green=5;loop"
signal.control(control_policy);

// later set to red in real-time
control_policy = "red=5;loop"
signal.control(control_policy);

..
```

gyp199009, 2020-06-02:
I don't get it, how to excute the following lines after the simulation is running ( sim.run() is excuted ) :
// set to green in real-time
control_policy = "green=5;loop"
signal.control(control_policy);

// later set to red in real-time
control_policy = "red=5;loop"
signal.control(control_policy);

..

ntutangyun, 2020-06-02:
@gyp199009 check out the following code snippet of PythonAPI

```python
// PythonAPI/lgsvl/simulator.py

...

@accepts((int, float), (int, float))
def run(self, time_limit = 0.0, time_scale = None):
    self._process("simulator/run", {"time_limit": time_limit, "time_scale": time_scale})

...

def _process(self, cmd, args):
    j = self.remote.command(cmd, args)
    while True:
        if j is None:
            return
        if "events" in j:
            self._process_events(j["events"])
            if self.stopped:
                break
         j = self.remote.command("simulator/continue")

```

Note that the python script does not halt after calling run(), instead,  it continues to communicate with the simulator and tries to handle events from simulator with `_process_events()` function. 

As a result, you could still send commands to the simulator when the simulator is running.

As for how to implement, if you know when to call the `signal.control(control_policy);`, you could try something like `sim.run(5);  signal.control(control_policy); sim.run() ...`

or you could use callbacks to execute `signal.control(control_policy);`

there are a bunch of callbacks for the agents (Ego, Ped, NPC) as described [here](https://www.lgsvlsimulator.com/docs/python-api/#agent-callbacks).

Hope it helps.

gyp199009, 2020-06-03:
I got what you meant, thank you!
But actually I want to have the traffic light in the simulator represent the real one in real world (means that the traffic light in simulator should recieve signals sent by the physical world in real-time) , is this possible?

ntutangyun, 2020-06-03:
@gyp199009, Of course, it's possible. You may need some code to receive the signal status from the real world, and pass this data to the `signal.control()`.

EricBoiseLGSVL, 2020-06-18:
@ntutangyun thanks for the help.

closed 2020-06-18 by EricBoiseLGSVL