lgsvl/simulator #778: How to create 360 camera and take screenshots with coordinates ?

ananas-jpg, 2020-05-30:
Hello,
I recently started working in your program and don't have any skills in any other. I have a couple of questions that I don't understand how to solve.

1.I need to make a 360 camera but I don't really understand what xi, cubemapsize values I should set? 2.I need to take a screenshots or something like that of the intersection with the coordinates how I can do it ?

EricBoiseLGSVL, 2020-06-01:
@ananas-jpg @rongguodong can help explain 360 camera settings.

rongguodong, 2020-06-01:
The `xi` parameter is for fisheye camera. It is not related to 360 camera.
The easiest way to create a 360 camera is to utilize Unity's Camera.RenderToCubemap and RenderTexture.ConvertToEquirect. You can build your own 360 camera sensor. In CameraSensorBase, after [we call Camera.RenderToCubemap](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/CameraSensorBase.cs#L373), the result is distorted by calling LensDistortion.UnifiedProjectionDistort. You can simply delete the distortion line and add a call to RenderTexture.ConvertToEquirect. Attached is the result I got for your reference.
[image]
As you can see in this screenshot, the six faces from the cubemap render result are not continuous. This is because of the auto exposure (since each face has different average lightness). We are still investigating this to find a solution. If you have a solution, you are welcome to create a fix via pull request.

ananas-jpg, 2020-06-01:
@rongguodong I'm sorry what about parametrs for call? 
` RenderTexture.ConvertToEquirect(CubemapTexture, Camera.MonoOrStereoscopicEye.monoscopic);`  
I tried this but it doesn't work.

EricBoiseLGSVL, 2020-06-01:
@ananas-jpg Have you used our camera sensor with 360 capabilities or are you creating a new sensor?

ananas-jpg, 2020-06-01:
@EricBoiseLGSVL I used your camera sensor with 360 capabilities. Just deleted `LensDistortion.UnifiedProjectionDistort` and added `RenderTexture.ConvertToEquirect(CubemapTexture, Camera.MonoOrStereoscopicEye.monoscopic);`

rongguodong, 2020-06-01:
> @EricBoiseLGSVL I used your camera sensor with 360 capabilities. Just deleted `LensDistortion.UnifiedProjectionDistort` and added `RenderTexture.ConvertToEquirect(CubemapTexture, Camera.MonoOrStereoscopicEye.monoscopic);`

It should be CubemapTexture.ConvertToEquirect(EquirectTexture, Camera.MonoOrStereoscopicEye.Mono). The first parameter is the output texture (a 2D texture, not a cubemap one), and you should create it yourself in the init stage.

ananas-jpg, 2020-06-01:
@rongguodong so I need to do a Cubemap conversion to a 2d texture? sorry I'm not very good at this, it must be something like this? If you can give me instructions or advice on how to implement this correctly please.
```
RenderTexture.active = SensorCamera.targetTexture;

            Texture2D tex2D = new Texture2D(RenderTexture.active.width, RenderTexture.active.height);
            tex2D.ReadPixels(new Rect(0, 0, RenderTexture.active.width, RenderTexture.active.height), 0, 0);
            tex2D.Apply(false);
```

rongguodong, 2020-06-02:
ConvertToEquirect is a member function of class RenderTexture, and CubemapTexture is a RenderTexture (created [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/CameraSensorBase.cs#L309)). So you can call CubemapTexture.ConvertToEquirect. To be able to call this function, the texture dimension must be of type TextureDimension.Cube. More details can be found at: https://docs.unity3d.com/ScriptReference/RenderTexture.ConvertToEquirect.html

You need to manually create an output texture with texture dimension TextureDimension.Tex2D. This online example may be helpful to you: https://blogs.unity3d.com/2018/01/26/stereo-360-image-and-video-capture/

ananas-jpg, 2020-06-02:
@rongguodong I don't understand what is parameter "equirect" and where i get it ?
cubemapRightEye.ConvertToEquirect(equirect,Camera.MonoOrStereoscopicEye.Right);

rongguodong, 2020-06-02:
You need to create it yourself. Here is another more complete example:
https://gist.github.com/MarcAlx/ee2cdfd6881ccc1faf70eaec3a191490

ananas-jpg, 2020-06-02:
@rongguodong I'm completely confused. if you can attach the code that gives you the result as shown in the screenshot.

rongguodong, 2020-06-02:
OK. Here you are:
[CameraSensorBase.zip](https://github.com/lgsvl/simulator/files/4718565/CameraSensorBase.zip)

ananas-jpg, 2020-06-04:
@rongguodong Thank you very much. If you sent me the correct code, then the problem is clearly not in it. I use the PythonAPI to run It, but maybe I'm wrong ? Should I need to use `CubemapSize` and `Xi` ? Attached my camera setting in JSON.
`{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width":  1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Fisheye": true, "Topic": "/apollo/sensor/camera/front_6mm/image/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},`
I tried putting  `CubemapSize` and `Xi` but result still doesn't match.

rongguodong, 2020-06-04:
What do you mean by "result still doesn't match"? 
In this hack, you do not need to set CubemapSize and Xi in JSON.

ananas-jpg, 2020-06-05:
@rongguodong I mean the picture I got on main camera doesn't match picture you sent

rongguodong, 2020-06-05:
Can you post what you get?

ananas-jpg, 2020-06-05:
@rongguodong 
[image]
How I call It
`import lgsvl
sim = lgsvl.Simulator(address = "localhost", port = 8181)
sim.load(scene = "SingleLaneRoad")
ego = sim.add_agent(name = "Lincoln2017MKZ (Apollo 5.0)", \
agent_type = lgsvl.AgentType.EGO, state = None)
sim.run()`
Looks like not all parts are connected

rongguodong, 2020-06-05:
@ananas-jpg You are not using Fisheye. To use Fisheye, you need both `Distorted` and `Fisheye` to be `true`. Please add `Distorted` in your JSON and try again.

ananas-jpg, 2020-06-05:
@rongguodong 
[image]
 it hates me:)

rongguodong, 2020-06-05:
@ananas-jpg Can you tell more details on how did you use my hack code? Did you do the following?
* Clone latest source code
* Replace CameraSensorBase.cs with my hack
* Run in Editor in API mode
* Run your Python script

ananas-jpg, 2020-06-05:
@rongguodong I downloaded the new release (2020.05) and modify the CameraSensorBase.cs. It works thank you !

ananas-jpg, 2020-06-06:
@rongguodong Is it possible to drive/control a vehicle using API?
 San Francisco doesn't work well.
[image]

EricBoiseLGSVL, 2020-06-08:
@ananas-jpg This looks like the simulator environmental effects shader is not compiled.  With any updates to lgsvl simulator, I recommend deleting the Library folder in your root project so Unity can re-compile shader graphs.  Also, be sure that screen shots shows the errors that you see in the Console panel or post the player log.  This will help to solve the issues that you may have.

ananas-jpg, 2020-06-08:
@EricBoiseLGSVL another question. where can I find a file with information about the position of the vehicle and how can I record it?

rongguodong, 2020-06-08:
If you mean the position of the ego car, you can find it in GpsSensor or ImuSensor.

closed 2021-07-16 by EricBoiseLGSVL

selfsimulator, 2021-08-10:
> 如果您指的是自车的位置，您可以在 GpsSensor 或 ImuSensor 中找到它。
你好，我也在研究LGSVL，请问能给个vx方便请教你一些问题嘛？

selfsimulator, 2021-08-10:
> @EricBoiseLGSVL另一个问题。在哪里可以找到包含车辆位置信息的文件以及如何记录？

你好，我也在研究LGSVL，请问能给个vx方便请教你一些问题嘛？

EricBoiseLGSVL, 2021-08-10:
@selfsimulator please post in English