lgsvl/simulator #781: Unable to setup radar with ROS1

MaxandreOgeret, 2020-06-01:
Hello, 

I am trying to setup the radar with LGSVL in Docker and I can't manage to make the Radar work, I get this error in the web interface : 

[image]

Thanks for your help

EricBoiseLGSVL, 2020-06-01:
@daviduhm Do know why the docker sensor is failing?  @MaxandreOgeret What version of simulator are you using?

rongguodong, 2020-06-01:
@MaxandreOgeret Could you please provide more details of how to reproduce this? What docker are you using? Which version of simulator are you using? Are you following our instructions here: https://www.lgsvlsimulator.com/docs/autoware-instructions/?

martins-mozeiko, 2020-06-02:
Radar implementation in Simulator supports only Apollo. To work with Autoware/ROS you'd need to add extra code to convert data produced by Radar sensor into whatever message ROS expects. You can see example, how this data is converted for Apollo here: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosConversions.cs#L151

EricBoiseLGSVL, 2020-10-05:
@MaxandreOgeret Same with your CanBus issue.  Look how the other sensors are supported and add the converting code to simulator source. #747

closed 2020-10-20 by EricBoiseLGSVL