lgsvl/simulator #782: Can not run standalone executable file

DmitryRedin, 2020-06-01:
I downloaded the file from the link https://github.com/lgsvl/simulator/releases/tag/2020.03
I launched the terminal in Ubuntu 16.04.
sudo chmod a+x ./simulator
Then in terminal i start simulator: ./simulator
But i can not run file "simulator".
[image]

dr563105, 2020-06-01:
Hey!
I use Ubuntu 18.04. I just tried to run the simulator from the terminal. It runs fine. I didn't change the permission nor run it in sudo mode. 

Did you execute this command before launching the simulator `sudo apt install libvulkan1`?
Most often if you install libvulkan, the simulator will launch. 

Please try that.

DmitryRedin, 2020-06-01:
@dr563105 
> Hey!
> I use Ubuntu 18.04. I just tried to run the simulator from the terminal. It runs fine. I didn't change the permission nor run it in sudo mode.
>
> Did you execute this command before launching the [...]

Thanks. I can start simulator, but i  can not see drop-down menu at the top of the screen.

[image]
 I want to create HD map for simulation.

dr563105, 2020-06-01:
I'm sorry I don't understand what you mean by "drop down menu". Do you mean simulator drop down to quit the application? or something else?

For HD map creation, you would need Unity. Have a look at this page --  https://www.lgsvlsimulator.com/docs/map-annotation/

You can find other helpful information in there as well -- https://www.lgsvlsimulator.com/docs/

DmitryRedin, 2020-06-01:
@dr563105 
> I'm sorry I don't understand what you mean by "drop down menu". Do you mean simulator drop down to quit the application? or something else?
>
> For HD map creation, you would need Unity. Have a look at [...]

[image]

dr563105, 2020-06-01:
Well that is screenshot of unity menu. The standalone executable is the result of built simulator from Unity. You can't find such drop down in the built version. 

You can run also the simulator from Unity itself. For that you need the cloned version of the simulator.

DmitryRedin, 2020-06-01:
@dr563105, thanks. Can i load point cloud and image to editor HD Map? Can i export and import vector map data (csv files for Autoware and xodr file to OpenDrive) to/from editor?

dr563105, 2020-06-01:
I don't know about those. Perhaps someone from the LGSVL team can tell you.

EricBoiseLGSVL, 2020-06-01:
@dr563105 Thanks for the help.  @DmitryRedin Yes, you can load HD map in the editor.  You can edit, import and export HD map data.  #775 has a link to a tutorial to load a scene with point cloud data.  You will need the Editor to build the point cloud scene.  Then, you can run an executable and add the path to the new point cloud map.

Malathi15, 2020-06-02:
Hi @dr563105 
I am using Ubuntu 18.04.
1) I have downloaded the Simulator from this link https://github.com/lgsvl/simulator/releases/tag/2020.03
2) After that in terminal I have run the following commands
sudo apt install libvulkan1
sudo chmod a+x ./simulator
./simulator
After that I have open the browser and select the vehicle and map. When I have run the simulator there is no output disply( Please check the screenshot)
[image]
[image]

Please help me to slove this problem

Thanks in Advance
Malathi K

dr563105, 2020-06-02:
Hey @Malathi15 !

Is this first time using the LGSVL simulator? If so you should see a few simulation setup already present. Like so with Valid status(ignore ROS2 Lane Following) 
[image]

Did you modify them? 

You don't need to select vehicle or map. Just go to Simulations tab, select one of those except Python API and run it. 

You can of course try deleting all config files. https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located. Please delete the 'LGSVL Simulator' directory.

Then run the standalone exe, open browser, wait till all the maps, vehicles get downloaded and then select a simulation.

DmitryRedin, 2020-06-02:
@dr563105 , @EricBoiseLGSVL , thank you for answers.

closed 2020-06-09 by EricBoiseLGSVL