lgsvl/simulator #783: About HD map

DmitryRedin, 2020-06-02:
Hello, i have some questions about HD map.
1. What information do hd maps store? 
Vector map(information about lane, traffic light, stopline, etc.)
Point Cloud Data (x,y,z data)
Aerial photo from drone or plane(raster data and georeference)
Mesh( textured terrain model)
2. I have to upload HD map to your server to run the simulation or I can store data on the local machine?

ntutangyun, 2020-06-02:
hi @DmitryRedin
1. basically you can store anything that you believe helpful for your autonomous driving functions in the HD maps. 
You could checkout [openDRIVE](https://www.asam.net/standards/detail/opendrive/) as an example. 

2. You could run the simulation locally by downloading the [latest build](https://github.com/lgsvl/simulator/releases/tag/2020.03) of the LGSVL Simulator.

AlexandrZabolotny, 2020-06-05:
@EricBoiseLGSVL , editor gizmos is active

[image]

I load point cloud and i try to create lane linked to point cloud. But i do not see any waypoint. I am following documentation.

[image]

I use Ubuntu 16.04

EricBoiseLGSVL, 2020-06-05:
Have you created the mesh collider?  Also, to turn on gizmos, please click the button, not the drop down.
Also, you need to show the Hierarchy Panel when taking screen shots or we have no idea how the scene is setup.  We don't need the Annotation panel open.

Here is another issue that I made some screen shots to help #791