lgsvl/simulator #786: Cannot find borregasAve in unity hub

saruvora, 2020-06-03:
Hi, I have finished till step 11 from this [link](https://www.lgsvlsimulator.com/docs/build-instructions/) But I am not able to find borregasAve in unity hub. I am using ubuntu 16.04. 
I have a few more questions(I am a beginner for unity and docker):
1. what do I need to install to get access to create my own maps and get position of the vehicles? 
(even for windows what do I need to install for this)
2. can I set a path for the vehicles to move in the simulation?

dr563105, 2020-06-03:
Hey!
If I'm not wrong BorregasAve must be loaded as an asset -- https://www.lgsvlsimulator.com/docs/assets/#adding-an-asset and https://content.lgsvlsimulator.com/maps/borregasave/ before you can see it in Unity. Have you done this already? 

To answer your questions -
1. To create maps you can see these instructions -- https://www.lgsvlsimulator.com/docs/map-annotation/
To get vehicle's position - you can use the sensors available and choose an appropriate bridge type to your program -- ROS, CyberRT. This part should be independent of the OS platform. 
2. I'm not an expert on this but I think you can achieve that by following steps similar as mentioned here -- https://www.lgsvlsimulator.com/docs/npc-map-navigation/

May be somebody from the LGSVL team can help you or correct me.

EricBoiseLGSVL, 2020-06-03:
@saruvora Hello and welcome to LG SVL Simulator.

1.  To find borregas ave and all other assets we provide as open source, please look at number 14 of the guide.  [Here](https://www.lgsvlsimulator.com/docs/assets/) is the link to the guide for adding assets.
2. [Here](https://www.lgsvlsimulator.com/docs/map-annotation/) is the link for map annotation.  You can create, import and export hd maps in multiple formats.  This annotation is what the NPC's follow.

I hope this helps.  If you are new to Unity and development, I recommend watching some YouTube videos on working in Unity.  It will help immensely.  Map creation and map annotation are advanced feature set of LG SVL Simulator so feel free to post any issues or questions here because it will help others that might have the same problems.

saruvora, 2020-06-04:
Great thank you to both of you :) and also for such a quick response. The asset part worked. Will see some videos before proceeding 
Thank you

closed 2020-06-04 by EricBoiseLGSVL