lgsvl/simulator #788: Shader Error, during unity project build

AlexandrZabolotny, 2020-06-04:
Hello!
Step 20 from [documentation](https://www.lgsvlsimulator.com/docs/build-instructions/).
We check Environments, Vehicles, choose folder, then click build button, then we get error in console

_Shader error in 'DiffuseDenoiser.compute': failed to open source file: 'Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/Raytracing/HDRayTracingManager.cs.hlsl' at kernel BilateralFilterColor at Packages/com.unity.render-pipelines.high-definition/Runtime/RenderPipeline/Raytracing/Shaders/ShaderVariablesRaytracing.hlsl(1) (on d3d11)_

[image]

AlexandrZabolotny, 2020-06-05:
@EricBoiseLGSVL,
Ubuntu 16.04
How can i check what i am using(Vulcan or OpenGL)?
[image]

Simulator release 2020.03

[image]

How can i check d3d11 support?

lemketron, 2020-06-05:
> @EricBoiseLGSVL,
> Ubuntu 16.04
> How can i check what i am using(Vulcan or OpenGL)?

You can search for "Vulcan" in `.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log`. 

You should see something like the following:
```
Vulkan API version 1.1.0 (1.1.119 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce RTX 2070 with Max-Q Design] id=[1f10]
```

> Simulator release 2020.03
> How can i check d3d11 support?
That's a Windows thing but will indicate whether your GPU has the needed hardware support; knowing which GPU you have would be helpful -- and that should appear as the Vulkan renderer in Player.log as noted above.

AlexandrZabolotny, 2020-06-05:
@lemketron,
```
Vulkan API version 1.1.0 (1.1.95 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce RTX 2070] id=[1f10]
Vulkan device type 2
```
Vulkan driverversion=[0x6895c040] uint=[1754644544]
Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]

EricBoiseLGSVL, 2020-06-08:
@AlexandrZabolotny That looks fine.  Did you delete the Library folder and re-compile shader code?

closed 2020-10-20 by EricBoiseLGSVL

Tufis16, 2024-08-20:
I see error from Book of the Dead demo:

Shader error in 'Shader Graphs/BotD_Graph_Lit_TranslucentAlphaCutoff': call to 'tex3D' is ambiguous at Assets/BookOfDeathAssets/Other/BotD/BotD_Common_ShaderGraph.hlsl(74) (on d3d11)