lgsvl/simulator #790: Sensors

AlexandrZabolotny, 2020-06-05:
Hello!
How can i get and build lidar, radar, camera and GPS?

[image]

ntutangyun, 2020-06-05:
You can check out the codes for those sensors at `Assets/Scripts/Sensors/`.

Customized sensors placed at `Assets/External/Sensors` will be listed in the **Sensors** section in your screenshot. Check out docs [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/)

rongguodong, 2020-06-05:
This build dialogue only lists "Sensor Plugins". Those you mentioned are build-in sensors which are built with the simulator itself.

EricBoiseLGSVL, 2020-06-05:
@AlexandrZabolotny @ntutangyun and @rongguodong are right.  There are two types of sensors in the simulator.

1) Built-in sensors that we provide with the source code, e.g., LiDAR, GPS, etc..  These are in Assets/Scripts/Sensors, Assets/Prefabs/Sensor and finally added to Assets/Resources/RuntimeSettings in SensorPrefabs Array. You can directly edit or duplicate these sensors to your needs. *NOTE* This can cause issues if you pull our latest release so we created Sensor Plugins to mitigate.

2) Sensor Plugins are what you see in Build... windows.  Like @ntutangyun pointed out, these need to be in the External folder to show up to build.  This helps keep your new sensors separated from source code.  We have [ComfortSensor](https://github.com/lgsvl/ComfortSensor) as an example.

closed 2020-06-09 by EricBoiseLGSVL