lgsvl/simulator #791: Creating of new HD Map

DmitryRedin, 2020-06-05:
Hello.
I use Windows 10.
I try to create lane, but i can not create waypoint.
Please see video https://drive.google.com/file/d/1Br998Ax6u79S1Nyt8ZCBqKQCMDb6heQd/view?usp=sharing

EricBoiseLGSVL, 2020-06-05:
@DmitryRedin Be sure to turn on gizmos in Scene View Panel, or you won't see the annotation objects that are created.  You also are not parenting the annotation correctly in the MapHolder object.  You need to place LaneSections as a child object under MapZuzino (This should be capitalized btw).  You also need to be closer to the ground collider to see the annotation gizmos, they are not large.  Here are some images to help.  Please follow our documentation on this [here](https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map).

Create the holder objects in the correct place and connect to the correct Map naming convention and MapHolder.cs references.  Create the LaneSection annotation
[image]

Again, be sure gizmos are turned on and set the new LaneSection as the parent
[image]

Create the temp waypoints, you can see them in the Hierarchy panel
[image]

Select how many waypoints you want in the straight lane and click the Connect button.  This removes the latest temp waypoints and creates the map lane.
[image]

Be sure to create a prefab of the map annotation and save in the correct places in the Assets/External/Environments/Zuzino folders.  Also be sure the top level parent holders Intersections and Lane Sections are set in MapHolder.cs references.
[image]

DmitryRedin, 2020-06-08:
@EricBoiseLGSVL, thank you.
I do not understand last action: _**Be sure to create a prefab of the map annotation and save in the correct places in the Assets/External/Environments/Zuzino folders. Also be sure the top level parent holders Intersections and Lane Sections are set in MapHolder.cs references.**_
I can not export HD map. I see error: **_ArgumentNullException: Value cannot be null.
Parameter name: key_**
Please see video: https://drive.google.com/file/d/12giA-3zuQ9f0zHILT_GX1uO4vQsbGMJr/view?usp=sharing

EricBoiseLGSVL, 2020-06-08:
@DmitryRedin I see that at 01:00 you drag the map annotation holders out of MapHolder parent.  This is wrong.  You rename the MapHolder object to Map[YourMapName] and create annotation as children of these holder objects.  If you save your scene and open the annotation tool, you can just click create Mapholder and it will do this automatically for you and delete the initial generated one.

Lanes and line objects go with LaneSections and Intersections have stop lines/lanes and lines with map signal or sign data.  After you are finished, you make a prefab of the Annotation data for source control and best practices working in Unity.  Point Cloud data should not be included in annotation data.  Please open the CubeTown map and look at how the annotation is organized.  This gives examples of Stop sign intersections, Traffic signal intersections and lanes.

DmitryRedin, 2020-06-09:
@EricBoiseLGSVL, thank you.
_**If you save your scene and open the annotation tool, you can just click create Mapholder and it will do this automatically for you and delete the initial generated one.**_

Where is "create Mapholder" button? I do not see this button in Annotation HD Map dialog.
============================================================

Sometimes i can not create HD Map, button Create Lane does not work.
Please see video: https://drive.google.com/file/d/1cWVltg6_erVZD4VY-r38qThuEiDpNU_q/view?usp=sharing

eyadshak, 2020-06-10:
The option to create a map holder does not appear for me if there is a holder already in the scene, it only appears if I delete the default holder.

Regarding the "Create Lane Section", I am also facing this issue. No object gets created in the scene hierarchy tab when I click on create lane section or intersection.

EricBoiseLGSVL, 2020-06-11:
@DmitryRedin Try this, delete the MapHolder object.  Then in the annotation tool it should have create map holder button on the top.  You won't see this button unless a map holder is missing from the scene.  

The reason you are not getting a lane section is because the ray casting to the collider is not working.  I'm not sure why this is happening.  Can you run the simulation and have a vehicle sit on the collider?  Can you show the collider in scene view during play mode?  The annotation tool needs to ray cast correctly to a collider before it can add any annotation objects.

If the map is flat you can place a large plane with a collider as a work around until we can figure out why the created collider is not working.

EricBoiseLGSVL, 2020-06-11:
@daviduhm Is there any settings in collider generation that would help?

DmitryRedin, 2020-06-16:
@EricBoiseLGSVL @daviduhm, I build colliders with the same parameters, at one time creating a lane section does not work, but sometimes it unexpectedly works.

EricBoiseLGSVL, 2020-06-16:
@DmitryRedin Please post an image of the collider you are making from the point cloud and trying to annotate.  Also, please show in the inspector panel the settings for the selected collider.  I have a feeling that there is an issue with the data and collider creation.

EricBoiseLGSVL, 2020-06-18:
@DmitryRedin You also may need to increase the detail level of the point cloud collider.  We are working on making this more option label more informative

DmitryRedin, 2020-06-19:
@EricBoiseLGSVL 
[image]
[image]

Size of ply file is 7.2Mb

System OS: 16.04
RAM Memory:  32 Gb
In terminal: lspci | grep -E "VGA|3D"
00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GK106GLM [Quadro K2100M] (rev a1)

EricBoiseLGSVL, 2020-06-19:
It looks like you have a bunch of holes in the collider, this could be why you cannot create a map annotation sometimes.  If there is no collider, the map annotation will not work.
- You need to increase the mesh detail level to 5-6
- I am worried that your GPU is not enough.  You need at least 8Gb GPU to run simulator

closed 2021-07-16 by EricBoiseLGSVL