lgsvl/simulator #792: Livox Lidar / Custom Lidar Patterns [Question]

osholsonel20, 2020-06-05:
Hey LGSVL just had a question in regards to modifying how the lidar creates a point cloud. Is there any effort in trying to recreate a Livox Lidar scan or a custom Lidar Scan? Otherwise I can always try and recreate it in my own sensor and leverage the Lidar Sensor script for the GPU async hand off optimization portion of it. Any tips or directions would be great!

https://www.livoxtech.com/mid-40-and-mid-100

rongguodong, 2020-06-05:
We do not support Livox Lidar now. However, the [Lidar Sensor Plugin](https://www.lgsvlsimulator.com/docs/lidar-plugin/) is there for users to create their own Lidar sensors matching their specific Lidar models. We have examples of Lidar Sensor Plugin for Velodyne VLP-16, VLP-32C, and VLS-128. You can refer to them and create Livox Lidar (and contribute here if possible).

osholsonel20, 2020-06-05:
Sounds good! Thanks for the heads up.