lgsvl/simulator #794: Is it real to control venicle on Windows by API?

eniseiskaya, 2020-06-06:
Hello
 Is it real to control venicle on Windows by API? I need to launch from unity and control the vehicle.

ntutangyun, 2020-06-07:
yes it's possible. Please check out the docs on [Python API](https://www.lgsvlsimulator.com/docs/python-api/)

eniseiskaya, 2020-06-07:
@ntutangyun For some reason, the example from the documentation did not work. I redid it and now it seems to be running. Thank you.
 Have you ever had such a bug? Is it possible to move the vehicle higher?
[image]

ntutangyun, 2020-06-08:
@eniseiskaya you could try setting the `z` in `AgentState` to a higher value when spawning the ego vehicle.

EricBoiseLGSVL, 2020-06-08:
@eniseiskaya Please see #778