lgsvl/simulator #796: Changing Horizontal FOV with a non-rotating Lidar Sensor

osholsonel20, 2020-06-09:
Hey LGSVL, I'm slowly working through the Lidar code, and I've got the rotation to stop, but when I try and change the HorizontalAngleLimit to a specific Angle that I want, it just breaks the whole thing and nothing outputs. Have any tips on navigating this code base? Basically, I'm trying to get a sensor that still does the rings that velodyne does, but in a targeted angle without rotation.

rongguodong, 2020-06-09:
@osholsonel20 Can you explain more details about what you are trying to do?
Why do you get the rotation stop? Do you want to simulate directional Lidar? Even for directional Lidar, it can still be implemented in the similar way by rotating the camera with a small FOV to cover its (directional) field of view.

The variable `HorizontalAngleLimit` defines the FOV of the camera. We calculate the angle covered by the Lidar within a certain frame (i.e. `Time.deltaTime * 360.0f * RotationFrequency`), and rotate the camera (with FOV of `HorizontalAngleLimit`) to cover this angle one by one. The rendering results are read back in parallel and the points are generated from the read-back results.
In the current implementation, the camera rotate to cover the full 360 degrees. But you should easily modify it if you only want part of it covered.

osholsonel20, 2020-06-09:
Yeah, so basically as you've seen before I'm trying to create the Livox Lidar. Based on some analysis i've figured out they are creating "sine" wave like patterns. My first thought was to retro fit the velodyne ring pattern, inside a specific area. As the Livox Lidar is not 360, I was going to filter it to a targeted angle, still with the ring pattern and go from there. Livox Lidar is 81.7 Horizontal FOV with a  25.1 Vertical FOV.

rongguodong, 2020-06-09:
Even with limited horizontal FOV (81.7 not 360), does it still scan this angle one column by one column? If so, our current implementation is still valid for this.

osholsonel20, 2020-06-09:
Kind of, I was breaking apart a bag file step by step, it has 12,000 points in one frame. Then the next frame it creates another 12,000 points on the opposite side to form a "full pointcloud". In one second of data it generates 240,000 points. Within these 12,000 points I've found 12 "sets" of 1000 points that contain either the top of the sine wave or the bottom. It's a really weird pattern and I haven't heard from Livox if they had any source on how it's generated.

Maybe keeping the ring pattern but rotating the camera in a wave inside of the specific FOV might work? It seems to "break up" the pointcloud when doing so, leaving gaps in between.

rongguodong, 2020-06-09:
Velodyne has a document explaining all details of how every point is generated. Without a document from the manufacture, we cannot simulate in the correct way. :)
Maybe you can contact Livox asking for similar document.

osholsonel20, 2020-06-09:
Sounds good, yeah I guess it's back to the reverse engineer drawing board, I'll have to do some mapping of output values and find some commonality between these infinity symbols it's drawing. Thanks for the help!

xmba15, 2020-07-10:
@osholsonel20 Have you succeeded in the reverse engineering target to simulate livox pointcloud? I am interested in "seeing the result" actually :).

here is the paper that I believe has relation to livox pointcloud pattern
https://arxiv.org/pdf/2006.11034.pdf

osholsonel20, 2020-07-10:
I have not sadly. This is a great resource though, thank you! This will just take lots of time, that I may want to spend elsewhere until we get the actual specs from the supplier. If I find anything though, i'll post back here!