lgsvl/simulator #798: AttributeError: 'LgsvlEnv' object has no attribute 'camera'

Malathi15, 2020-06-10:
Hi,
I have tried to execute gym-lgsvl following this guide https://github.com/lgsvl/gym-lgsvl.
When I have execute the following command 
`python -m baselines.run --alg=a2c --env=gym_lgsvl:lgsvl-v0 --num_timesteps=1e5`
 I got an Attribute Error 
`AttributeError: 'LgsvlEnv' object has no attribute 'camera'`
(check the screenshot)
[image]

Please help me to solve this

Thanks in Advance
Malathi K

daviduhm, 2020-06-11:
@hadiTab can you look into this?

hadiTab, 2020-06-13:
@Malathi15 what version of the simulator are you using?

The gym-lgsvl code expects you to be using the `Lexus2016RXHybrid (Autoware)` vehicle. Can you check the JSON configuration for that vehicle and make sure that it has a camera?