lgsvl/simulator #800: Problem about building and running from source

chen1qy, 2020-06-11:
Hi, I have some problem when building from source. 
[image]

And the simulator doesn't work:
[image]

the OS is Ubuntu16.04, and the project shows as
[image]

chen1qy, 2020-06-11:
Is the Unity ok?
[image]

EricBoiseLGSVL, 2020-06-11:
@chen1qy First off you only need Linux and Windows Support.  No WebGL and no need for documentation.  Check is fine, it is more for internal development anyway.
You need to post player log so we can see errors.  Also, what environment are you trying to build?

chen1qy, 2020-06-11:
I choose the SanFrancisco and Jaguar2015XE to play the simulator. The environment built is as follows:
[image]

Sorry, I don't find the player log. Maybe I should learn about unity first.

EricBoiseLGSVL, 2020-06-11:
@chen1qy Ok, thanks.  Yes, player log from binary or editor can be found from every run and is very helpful to find errors or issues.  Still, I think the issue you have is not installing git LFS before you cloned the scenes.  Please use git lfs install (download if you haven't first), if you have downloaded it then git lfs pull.  You should be able to see the files are textures/meshes etc not text files pointing to urls.  Also, try building one asset at a time, not all at once.  This can help narrow the issue maps/vehicles.

chen1qy, 2020-06-12:
Thanks very much for your reply. I will try it.

closed 2020-06-18 by EricBoiseLGSVL

devynliu, 2020-08-12:
hello,i also meet this question.How can you resolve it? thanks!

ntutangyun, 2020-08-12:
@devynliu have you used git lfs ?