lgsvl/simulator #801: AttributeError: 'EgoVehicle' object has no attribute 'on_comllision'

xiao-hua-sheng, 2020-06-11:
This problem occurs in the process of reinforcement learning.

Traceback (most recent call last):
  File "./random_agent.py", line 15, in <module>
    observation, reward, done, info = env.step(action)
  File "/home/yons/anaconda3/envs/RL_LGSVL/lib/python3.6/site-packages/gym/wrappers/time_limit.py", line 16, in step
    observation, reward, done, info = self.env.step(action)
  File "/home/yons/Project/RL_LGSVL/gym-lgsvl/gym_lgsvl/envs/lgsvl_env.py", line 83, in step
    self.ego.on_comllision(self._on_collision)
AttributeError: 'EgoVehicle' object has no attribute 'on_comllision'

ntutangyun, 2020-06-11:
looks like a typo:
it should be `on_collision` I guess

xiao-hua-sheng, 2020-06-11:
> 看起来像错字：
> 应该是`on_collision`我猜
This is an official document. I haven't revised it. Change to "on"_ Collision "is the same mistake

closed 2020-06-11 by xiao-hua-sheng