lgsvl/simulator #802: CANBus on ROS

tgif73, 2020-06-11:
Hello,
I've seen this question being asked multiple times and looks like canbus message is not supported in ros message format. Do you have any plan to publish canbus message in ros? If not, can you share how we can publish canbus message in ros msg format so I can subscribe to the message without installing Apollo.

Thanks

dr563105, 2020-06-11:
Hey!

I'm not sure if my answer is right. 

I think you need to go in the direction of creating custom sensor plugins. You can [this ](https://github.com/lgsvl/LaneFollowingSensor)and [this](https://github.com/lgsvl/ComfortSensor) to get started. And use [this](https://www.lgsvlsimulator.com/docs/sensor-plugins/) to build your Unity project. 

Hope this helps.

martins-mozeiko, 2020-06-11:
Right, adding custom sensor plugin for your needs is correct way to go if you want to maintain this sensor functionality outside of simulator code.

Alternative is to edit simulator existing code to add support for publishing CanBusData type to ROS environment, not just Apollo. Then you could submit pull request for us to include it main repository. The relevant code is here: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosBridge.cs#L267 As you can see it always converts CanBusData to Apollo specific Apollo.ChassisMsg type. To support ROS it needs some Ros message type to convert. See example how this is done a bit below - for ImuData type. ImuData data is converted either to sensor_msgs/Imu (ROS) or Apollo.Imu (for Apollo) type.

The problem of supporting canbus message type is that afaik there is no standard canbus ros message. That's why we could not include it by default.

tgif73, 2020-06-11:
Thank you @dr563105 @martins-mozeiko .

closed 2020-06-16 by EricBoiseLGSVL