lgsvl/simulator #803: Multiple Lidar data collection synchronization

SeymurSm, 2020-06-12:
Hi All,
We are using 4 lidars on ego vehicle. To collect data by using PythonAPI I have use 4 different threads to collect each lidar data. The problem is there is no synchronization between output data and it is being a problem when we try to merge data of lidars on a single point cloud. This difference diverges 100 , 150 ms on some lidars. I think it is because of both capture and save latency. Some threads runs slower than others. Is there any way to collect each lidar on the same time frame? It is also possible to solve this problem by stopping ego vehicle movement on ever capture. Is it possible by means of PythonAPI? 

Thanks!

rongguodong, 2020-06-12:
Currently we do not have any synchronization between sensors -- each Lidar sensor generates and publish points independently. If you use [Velodyne Lidar Sensor](https://www.lgsvlsimulator.com/docs/lidar-plugin/#velodyne-lidar-sensor-plugins), you will have timestamp for every point, which may be used to match points from different Lidar sensors.

EricBoiseLGSVL, 2020-06-12:
@SeymurSm It would be better to process the point cloud data from timestamps after collection like @rongguodong posted.  Velodyne Lidar Sensor would be best for this

SeymurSm, 2020-06-15:
Hi @EricBoiseLGSVL . I have changed some parts of simulator and PythonApi to collect merged lidar data at once. It is working but it is slow as expected. Below snapshot is from visualization of merged data.
[image]
. 
Could you please share the link that @rongguodong  posted this process.

EricBoiseLGSVL, 2020-06-15:
@SeymurSm Yes, we are saying that it would be better to do this after a simulation has run.  We can't share the code at this time but I'll see if we can point you in the right direction

EricBoiseLGSVL, 2020-06-18:
@daviduhm any pointers to merge multiple point clouds?

closed 2021-07-16 by EricBoiseLGSVL