lgsvl/simulator #804: Bridge seems to not work when running latest Apollo with LGSVL Simulator

AIasd, 2020-06-13:
Hi,
I follow [the document on the LGSVL website](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/) about Running latest Apollo with LGSVL Simulator. Everything seems to go well until when I come to the step of running `./scripts/bootstrap_lgsvl.sh`, It says `could not launch module monitor` (see the screenshot below). Despite this, I follow the rest of the instructions but it seems that apollo is not connected with LGSVL as shown in the screenshot below. When I try to "Enable Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control.", these buttons automatically "move back". Route Editing tab is also always grey.

[image]

I will highly appreciate if anyone can shed some light on the issue I am encountering.

ntutangyun, 2020-06-13:
you could try to launch the monitor separately in a terminal

```bash
# get into Apollo docker container
./docker/scripts/dev_into.sh

# start monitor manually
cyber_launch start modules/monitor/launch/monitor.launch
```

and see if you could run the `monitor` manually without errors.

AIasd, 2020-06-13:
@ntutangyun I encountered the following error after manually launching the monitor separately in a terminal:
```
[cyber_launch_7048] INFO Launch file [/apollo/modules/monitor/launch/monitor.launch]
[cyber_launch_7048] INFO ========================================================================================================================
[cyber_launch_7048] INFO Load module [monitor] library: [monitor] [CYBER_DEFAULT] conf: [/apollo/modules/monitor/dag/monitor.dag] exception_handler: []
[cyber_launch_7048] ERROR Subprocess Popen exception: [Errno 2] No such file or directory: 'mainboard': 'mainboard'
[cyber_launch_7048] ERROR Start manager [monitor] failed. Stop all!
Traceback (most recent call last):
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 528, in <module>
    main()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 514, in main
    start(params.file)
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 444, in start
    all_died = pmon.run()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 285, in run
    p.get_exit_state()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 200, in get_exit_state
    if self.popen.returncode is None:
AttributeError: 'NoneType' object has no attribute 'returncode'
[cyber_launch_7048] INFO All processes have been stopped.
```

ntutangyun, 2020-06-13:
@AIasd have you successfully built everything with `bash apollo.sh build_gpu`?

lemketron, 2020-06-13:
> @AIasd have you successfully built everything with `bash apollo.sh build_gpu`?

It sounds like apollo is not built yet. But as an aside I would recommend building with `bash apollo.sh build_opt_gpu` since that will build optimized code that should perform better. I recently updated the LGSVL documentation to suggest this instead of "build_gpu".

AIasd, 2020-06-13:
Hi @ntutangyun and @lemketron ,
I actually built apollo with "Build passed". The following is the full log of building apollo, running it and running monitor separately.
```
zhongzzy9@in_dev_docker:/apollo$ ./apollo.sh build_opt_gpu
System check passed. Build continue ...
[WARNING] ESD CAN library supplied by ESD Electronics does not exist. If you need ESD CAN, please refer to third_party/can_card_library/esd_can/README.md.
Running build under GPU mode. GPU is required to run the build.
[INFO] Start building, please wait ...
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
.
____Loading package: modules/localization/msf/local_map/lossless_map
____Loading package: modules/perception/fusion/lib/dummy
____Loading package: modules/perception/onboard/transform_wrapper
____Loading package: modules/prediction/scenario
____Loading package: modules/perception/common/io
____Loading package: modules/canbus/vehicle/gem/protocol
____Loading package: modules/prediction/evaluator/vehicle
____Loading package: modules/planning/tasks/deciders/speed_bounds_decider
____Loading package: modules/perception/common
____Loading package: cyber/message
____Loading package: cyber/tools/cyber_monitor
[INFO] Building on x86_64...
[INFO] Building with --jobs=28 --ram_utilization_factor 80 for x86_64
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: (06-13 03:46:09.650) Found 4507 targets...
Slow read: a 2083539088-byte read from /apollo/.cache/bazel/540135163923dd7d5820f3ee4b306b32/external/pytorch_gpu/lib/libtorch.so took 10702ms.
Slow read: a 2083539088-byte read from /apollo/.cache/bazel/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-opt/bin/_solib_k8/_U@pytorch_Ugpu_S_S_Cpytorch_Ugpu___Uexternal_Spytorch_Ugpu_Slib/libtorch.so took 9640ms.
INFO: (06-13 03:46:30.997) Elapsed time: 25.025s, Critical Path: 9.71s
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: (06-13 03:46:33.951) Found 4507 targets...
INFO: (06-13 03:46:34.211) Elapsed time: 1.996s, Critical Path: 0.00s
1
==============================================
[ OK ] Build passed!
[INFO] Took 41 seconds
==============================================
zhongzzy9@in_dev_docker:/apollo$ ./scripts/bootstrap_lgsvl.sh
nohup: appending output to 'nohup.out'
nohup: appending output to 'nohup.out'
Could not launch module monitor. Is it already built?
nohup: appending output to 'nohup.out'
Launched module dreamview.
Dreamview is running at http://localhost:8888
zhongzzy9@in_dev_docker:/apollo$ cyber_launch start modules/monitor/launch/monitor.launch
[cyber_launch_8134] INFO Launch file [/apollo/modules/monitor/launch/monitor.launch]
[cyber_launch_8134] INFO ========================================================================================================================
[cyber_launch_8134] INFO Load module [monitor] library: [monitor] [CYBER_DEFAULT] conf: [/apollo/modules/monitor/dag/monitor.dag] exception_handler: []
[cyber_launch_8134] ERROR Subprocess Popen exception: [Errno 2] No such file or directory: 'mainboard': 'mainboard'
[cyber_launch_8134] ERROR Start manager [monitor] failed. Stop all!
Traceback (most recent call last):
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 528, in <module>
    main()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 514, in main
    start(params.file)
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 444, in start
    all_died = pmon.run()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 285, in run
    p.get_exit_state()
  File "/apollo/cyber/tools/cyber_launch/cyber_launch", line 200, in get_exit_state
    if self.popen.returncode is None:
AttributeError: 'NoneType' object has no attribute 'returncode'
[cyber_launch_8134] INFO All processes have been stopped.
```

rongguodong, 2020-06-13:
I got exactly same errors after pulled latest master. It seems nothing related to LGSVL. Can you just run normal bootstrap.sh? If not, please create an Apollo issue.

BTW: I tried manually append "/apollo/bazin-bin/cyber" to $PATH, and then I can run bootstrap and launch modules.

lemketron, 2020-06-13:
Strange, my (slightly out of date) apollo-master build has /apollo/bazel-bin/cyber in the $PATH; I'm curious why that wouldn't be getting added into $PATH in newer builds?

lemketron, 2020-06-13:
Try checking out https://github.com/ApolloAuto/apollo/commit/99135e093eed40957ca57e5e3dfc61e8494da5d5 (currently just two changes from tip of master) as this looks like it was broke in a very recent change.

lemketron, 2020-06-13:
bootstrap_lgsvl.sh and bridge.sh both work fine at that (HEAD^^) change, btw: https://github.com/ApolloAuto/apollo/commit/99135e093eed40957ca57e5e3dfc61e8494da5d5

lemketron, 2020-06-13:
Apollo master should be fixed as of this change which resolves this PATH issue: https://github.com/ApolloAuto/apollo/pull/11458

closed 2020-06-13 by lemketron