lgsvl/simulator #807: why UI objects doesnt show up in unity

DanielChungYi, 2020-06-17:
Hi, can anyone tell me why my object in the scene looks like this?
[image]
The environment is download from LGSVL repository without any modification. I had change the mode to shaded mode but still can't see the UI object. Any solution? Thanks a lot!

EricBoiseLGSVL, 2020-06-17:
@DanielChungYi Hello.  There are quite a few issues with this.
- You have three scenes open at once.  You should have Loader and the environment scene.  Which one are you loading? DuckieDownTown or San Francisco?
- You have a Sedan mesh in the scene in San Francisco.  If this is static, it is fine but this is not how you add an EGO vehicle.
- You need to show the console log or player log so we can see the errors you are having.

Since this is in such an odd state I suggest we start with the basics.
- What is your PC specs?  I see Vulkan which is good
- What version of LGSVL Simulator are you running?  Latest is 2020.05
- What are you trying to accomplish?  This looks like SF and our old robot DuckieTown, which is not compatible with the current simulator.
- Documentation [here](https://www.lgsvlsimulator.com/docs/)
- You cannot run old simulator with new assets, these are HDRP only and will not work

DanielChungYi, 2020-06-18:
@EricBoiseLGSVL Thanks a lot for replying.
I am using Intel i9 9900K with GeForce GTX 1660. 
The version of simulator is the latest. What I'm trying to do is modify the scene I download from [here](https://github.com/lgsvl/SanFrancisco) to fit my experiment. Now I realize that the latest version of the simulator must be edited with HDRP assets. So the assets in [here](https://www.lgsvlsimulator.com/docs/assets/) don't support HDRP? Then where can I get the assets that work with the new simulator?

EricBoiseLGSVL, 2020-06-18:
@DanielChungYi 
- GTX 1660 is not enough to run simulator and an AD stack.  Minimum we recommend is 8GB GPU memory.
- No, those assets are fine and they run in the latest simulator
- Please watch [this](https://youtu.be/S0w2ahhEPbE) tutorial showing how to create a map
- It is fine to edit SF but why do you have DuckieDownTown.scene and Loader.scene in the Hierarchy?  This is not correct.  Multi-scene editing is not needed, nor is it recommended.
- To ''see" the model in scene view, just press the light icon, toggle off if in playmode and you want to "see" the model in scene view
[image]

DanielChungYi, 2020-06-24:
Thanks for the reply.
I had tried the light icon before, but it doesn't work. 
Do I have to change any other configuration setting in Unity to show the model? 
And if I want to modify the scene with the latest simulator, do I have to run it on Windows?

EricBoiseLGSVL, 2020-06-24:
@DanielChungYi Are the models actually there?  Did you clone with git lfs?  Look in the project files for the meshes that are in scene and screen shot the actual meshes please?  We support Windows and Linux Vulkan, not opengl

closed 2020-10-20 by EricBoiseLGSVL