lgsvl/simulator #808: Web UI Vehicles For CyberRT

yumianhuli2, 2020-06-17:
Hi ! There is an example for ROS configuration by JSON on https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-add-a-vehicle. For example, if I want to configure JSON for CyberRT and set it to autopilot mode, how could  I configure it?Thank U!

EricBoiseLGSVL, 2020-06-17:
@yumianhuli2 We have default vehicle configurations for simulator when you start the WebUI.  The Lincoln2017MKZ with Apollo 5.0

lemketron, 2020-06-18:
This is documented in [Launching Apollo alongside the Simulator](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#launching-apollo-alongide-the-simulator), and the sample sensor configuration (if you want to add it to a different vehicle that isn't already configured for use with Apollo) can be found [here](https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/).

yumianhuli2, 2020-06-18:
> @yumianhuli2 We have default vehicle configurations for simulator when you start the WebUI. The Lincoln2017MKZ with Apollo 5.0

Got it！

yumianhuli2, 2020-06-18:
> This is documented in [Launching Apollo alongside the Simulator](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#launching-apollo-alongide-the-simulator), and the sample sensor [...]

Got it！

closed 2020-06-18 by yumianhuli2

yumianhuli2, 2020-06-19:
> @yumianhuli2 We have default vehicle configurations for simulator when you start the WebUI. The Lincoln2017MKZ with Apollo 5.0

HI! Do you mean that there is no need to configure any vehicle in the webUI browser，The default is Lincoln2017MKZ? What are the communication methods and parameters by default?

I think is still need to download Lincoln2017MKZ vehicle model and configuration of the json https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/? But I have not found Lincoln2017MKZ on this page https://www.lgsvlsimulator.com/docs/assets/ and BorregasAve map on https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#launching-apollo-alongide-the-simulator.
So,where could I find these?Thank U!