lgsvl/simulator #809: Bridge does not connect correctly.

SungYinYang, 2020-06-17:
Hi, I got a problem similar to #804.

My bridge is not connecting correctly.
I have followed the latest tutorial and build Apollo successfully today.
It seems that the localization is always having an error.

[image]

Also, my cyber_launch start also has error messages.
[image]

Any help would be appreciated!

EricBoiseLGSVL, 2020-06-17:
@luqiang21 Any idea why this is happening?

rongguodong, 2020-06-17:
Are you using our fork of Apollo 5.0 or latest Apollo master (5.5)?
If later, please select the correct setup mode "Mkz Lgsvl".

rongguodong, 2020-06-17:
From the screenshot of Dreamview, I believe you are using Apollo master, but with a wrong setup mode. :)

SungYinYang, 2020-06-17:
Thanks! I would try it out.

lemketron, 2020-06-17:
Here's a screen shot that demonstrates the proper settings for Dreamview when using apollo-master...
[image]

SungYinYang, 2020-06-18:
@lemketron @rongguodong Thanks for your clarification!
I am able to see my Dreamview when using apollo master.

However, after I click Submit ROUTE in Dreamview, the car in the LGSVL did not move.
Any help would be appreciated.

Here are some settings of my Dreamview.

Layer Menu:
[image]

Module Controller:
[image]

lemketron, 2020-06-18:
The hourglass in the fence indicates that Apollo auto mode is "not ready" according to the [Dreamview documentation](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/dreamview_usage_table.md).

Sometimes it helps to "nudge" the car a little by hitting the up-arrow on the keyboard, but in general Apollo should start driving if everything is ready and it is communicating with the simulator (and the fact that localization has placed the ego vehicle properly on the Borregas map is a good sign).

SungYinYang, 2020-06-18:
@lemketron Thanks for your fast reply.
I would try it now!

SungYinYang, 2020-06-18:
@lemketron 
I have tried multiple times in running the same routes.
I succeeded to make the car move in auto mode by nudging the car a little bit by hitting the up array for the first time. 
However, all of my attempts failed after the first time.
It seems pretty not stable for me.

I notice that the "not ready" sign exists in my scenario forever.
May I know what apollo version you are using?
Do you recommend me to download the apollo 5.0 instead of the latest master?

Thanks again for all your reply.
I really appreciate your time in answering my new baby questions.

lemketron, 2020-06-18:
On what hardware are you running the Simulator and Apollo on and with which GPU?

If your system supports Apollo 5.0 I would recommend trying that instead, from the [LGSVL fork](https://github.com/lgsvl/apollo-5.0) as documented [here](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/).

In addition, if you are trying to run both apollo and simulator on a single machine that can sometimes be too demanding and it may help to run apollo on a separate machine (but then be sure to change the bridge address in the vehicle configuration from localhost to the IP address of the apollo machine where the CyberRT bridge is running).

SungYinYang, 2020-06-18:
I am running it with the Ubuntu system and with an RTX 2080S GPU.
Oh, I see that the requirement for Apollo 5.0 do not support my GPU.
Does my GPU support the latest Apollo (in the master)?

Currently, I am running both of them on the same machine.
Yeah, I notice that sometimes my machine would get overheated by using them together.
I would definitely put them separately later on.

Thanks again!

lemketron, 2020-06-18:
I have run apollo-master on a RTX-2070 (Razer Blade) laptop with the simulator running on the same machine (with Ubuntu 18.04) but it is definitely challenging. In particular, I usually close Chrome and anything else that might be using the gpu (see `nvidia-smi` tool for GPU memory usage). 

Otherwise, one of the apollo modules that use GPU (usually perception) can crash, which would definitely prevent the car from moving (but you would know as the switch will move back to "off" in dreamview modules view). 

Running dreamview in Firefox will use less GPU (than Chrome), and running it on a separate computer altogether will free up GPU and CPU resources.

If there isn't enough CPU performance available (see `top` tool), then it's possible that apollo may not reach the "ready" state to be able to drive. That said, I have run the simulator and apollo on a single machine with an 8GB GPU. Just don't leave anything else running that isn't needed. I do get better results running apollo 5.0 on a separate machine with a GTX-1080.

SungYinYang, 2020-06-18:
Thank you so much for all your help!
Your advice is very valuable to me!

SungYinYang, 2020-06-22:
@lemketron 

Hey, I got a follow-up question.
Currently, I am requesting another GPU machine so I am not able to put the two programs in two computers yet.

Recently, I have tried to build Apollo 5.0 on my RTX2080S GPU machine.
I found out that I was able to build it.
However, the result is quite similar to my Apollo5.5 one.
Both of them would not start driving after I click "submit route".
I have tried to nudge them a little bit.
The car would move, but the car did not follow the correct route.
The car would just go straight.

Is this a CPU resources problem?
Or is this a GPU problem that why RTX GPU card is not suitable for Apollo 5.0 as stated in LGSVL documentation?
[image]
I notice that I am not able to see the block of information in front of my ego vehicle in Apollo 5.0.

Thanks again for your time in answering all my previous questions!

luqiang21, 2020-06-22:
@SungYinYang Looks like your planning module is not working. The car is not getting control messages from Apollo. I think the car moves because you moved it a bit, then it continues. You need to check the module status on `Module Controller` tab to see if other modules are working correctly.

lemketron, 2020-06-22:
The RTX is not supported by Apollo 5.0.  If you want to run on RTX you should [build apollo master](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/).

SungYinYang, 2020-06-23:
Thanks for all of the answers!

SungYinYang, 2020-07-07:
@lemketron 

Thanks for your previous advice!
I have successfully connected two machines for Apollo5.5 (master) and LGSVL simulator.
Both machines used RTX 2080 GPU card.
The location on the dreamland would change when I move the car in the LGSVL simulator.

However, I am still not able to send a request route to run the EGO vehicle from point A to point B.
I notice that the auto mode is still not working.
[image]

I also notice that a few weeks ago, the picture that you send in this issue has the same problem.
[image]
(There is a white stop sign indicate that auto mode is not working.)

I am wondering that is there any other problem that would affect the auto mode status?
Since I have used two machines, I am pretty sure that it is not the problem of demanding CPU resources.

I have tried to use a "cyber monitor" to monitor my module.
I also try to execute every possible module in the apollo/scrips file.
The following is my cyber monitor.
[image]

I notice that there is a script "pad.sh", which is related to the auto-drive mode.
[image]

It asks me to enter 0 or 1 to either "reset to manually drive mode" or "auto-drive mode".
I tried to enter 1, but it does not help.

Any help would be appreciated!
Thanks!

rongguodong, 2020-07-07:
It seems your traffic light module is not running.

rongguodong, 2020-07-07:
What is that "pad_terminal"? Where do you enter 0 or 1?

SungYinYang, 2020-07-07:
@rongguodong 

Yes, I notice that my traffic light module cannot turn on. 

"pad.sh" is a script inside apollo/scripts. I think these scripts are related to the apollo module that is in the screenshot.
[image]

I enter 0 or 1 in the apollo container after I run the script.

rongguodong, 2020-07-07:
Do you see any errors in the log of traffic light module? Why cannot it run?
Do you use "bootstarp_lgsvl.sh" to start the dreamview? Do you have decompressed images in cyber_monitor?

What is "pad.sh"? Do you run it manually? I never used it before.

SungYinYang, 2020-07-08:
@rongguodong 

Do you see any errors in the log of traffic light module? Why cannot it run?
**1. I am not able to see traffic light log. May you tell me the file name of the log?
When I click the traffic light module in Apollo Dreamland, it would turn blue for 1 second and then turn off again.
** 

Do you use "bootstarp_lgsvl.sh" to start the dreamview? Do you have decompressed images in cyber_monitor?
**Yes, I did use "bootstarp_lgsvl.sh". I do not know that I have decompressed images in cyber_monitor or not.**

What is "pad.sh"? Do you run it manually? I never used it before.B
**I run it manually to see if this is the problem for my automode.
The problem that I encounter is that I am not able to start the automode to let my car to drive to a destination. 
The car never move by itself.**
**Is it possible for you to give us a demo video on how apollo drive automatically from point A to the selected destination? I think I have strictly followed the guideline on https://www.lgsvlsimulator.com/docs/apollo-master-instructions/ but we have not successfully run a single demo for weeks.  **
Thanks for your help again!

rongguodong, 2020-07-08:
1. All logs can be found in /apollo/data/log.
2. You should see channels like "/apollo/sensor/camera/front_6mm/image/compressed" which is compressed image published by the simulator, and "/apollo/sensor/camera/front_6mm/image" which is decompressed image published by Apollo's decompressing module. Traffic light perception needs decompressed images to work.

SungYinYang, 2020-07-09:
Hi rongguodong,

Thanks again for your answers.

I am not able to find logs related to what you mention.
I have searched both inside the container and outside the container.
(outside)
[image]
(inside)
[image]

I am not able to find channels like you mentions "/apollo/sensor/camera/front_6mm/image/compressed".
[image]

Is it possible for you to provide more details? A screenshot would be even better.

Also, do you know what modules are required to open for the automode?
The following images show that traffic light is not working.
[image]

Thanks again for all your help!

rongguodong, 2020-07-09:
@SungYinYang 

For perception log, you should have a "perception.INFO" under /apollo/data/log. Both obstacle and traffic light detection should write logs into it. Another way is to manually start traffic light perception by calling:
"cyber_launch start /apollo/modules/perception/production/launch/perception_trafficlight.launch"
You should see its log in the console.

To see Cyber channels, you need to run "cyber_monitor", instead of searching it in folders. Attached is a screenshot showing both "/apollo/sensor/camera/front_6mm/image/compressed" and "/apollo/sensor/camera/front_6mm/image" as green (i.e. published).
[image]

The required modules are clearly listed in [our instructions](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/#launching-apollo-alongide-the-simulator):
"Enable Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control"

closed 2020-10-20 by EricBoiseLGSVL