lgsvl/simulator #810: Cannot run the executable "Simulator"

ivanchristina, 2020-06-18:
Error: 
After clicking the simulator,  program crashed immediately.

What I do:
1. sudo add-apt-repository ppa:graphics-drivers/ppa
https://github.com/lgsvl/simulator/issues/782
result: 
Press [ENTER] to continue or Ctrl-c to cancel adding it.
......
Get:29 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2.464 B]
Fetched 8.647 kB in 4s (2.307 kB/s)                                 
Reading package lists... Done

2. sudo apt-get update
result:
Hit:1 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:2 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease                      
Hit:3 http://de.archive.ubuntu.com/ubuntu bionic InRelease                                       
Hit:4 https://download.docker.com/linux/ubuntu bionic InRelease          
Hit:5 http://de.archive.ubuntu.com/ubuntu bionic-updates InRelease       
Hit:6 http://de.archive.ubuntu.com/ubuntu bionic-backports InRelease
Reading package lists... Done 

3. trying to selecting Nvidia driver, but failed, as no Nvidia driver exits in additional driver tab in software & update. But I think the failed step should not result in "simulator" cannot run.

4. sudo apt install libvulkan1

result:
The following NEW packages will be installed:
  libvulkan1 mesa-vulkan-drivers
0 upgraded, 2 newly installed, 0 to remove and 101 not upgraded.
Need to get 2.348 kB of archives.
After this operation, 11,6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic/main amd64 libvulkan1 amd64 1.1.126.0-2~gpu18.04.1 [101 kB]
Get:2 http://de.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 mesa-vulkan-drivers amd64 19.2.8-0ubuntu0~18.04.3 [2.247 kB]
Fetched 2.348 kB in 1s (3.667 kB/s)                                          
Selecting previously unselected package libvulkan1:amd64.
(Reading database ... 164638 files and directories currently installed.)
Preparing to unpack .../libvulkan1_1.1.126.0-2~gpu18.04.1_amd64.deb ...
Unpacking libvulkan1:amd64 (1.1.126.0-2~gpu18.04.1) ...
Selecting previously unselected package mesa-vulkan-drivers:amd64.
Preparing to unpack .../mesa-vulkan-drivers_19.2.8-0ubuntu0~18.04.3_amd64.deb ...
Unpacking mesa-vulkan-drivers:amd64 (19.2.8-0ubuntu0~18.04.3) ...
Setting up libvulkan1:amd64 (1.1.126.0-2~gpu18.04.1) ...
Setting up mesa-vulkan-drivers:amd64 (19.2.8-0ubuntu0~18.04.3) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...

5. I refer to the other issues, but in my case libvulkan1 is installed.
[https://github.com/lgsvl/simulator/issues/782](url)

System:
Ubuntu 18.04 64-bit Virtual Box  Guest on Win 10 Host.
Graphic shows in virtual machine ubuntu: llvmpipe (LLVM 9.0, 256 bits)

dr563105, 2020-06-18:
Can you provide us with simulator player log? You can find the path to it in the FAQ in the lgsvl simulator docs. What is the output of `nvidia-smi` if any?

EricBoiseLGSVL, 2020-06-18:
@ivanchristina It seems you are trying to run this with VM.  This is not supported out of the box and you need a few steps to do this.  Simulator requires a GPU with minimum specs listed in the documentation.  We have a few issues covering this already but it is up to you to have GPU support for the simulator.   #708 is one example

ivanchristina, 2020-06-19:
> Can you provide us with simulator player log? You can find the path to it in the FAQ in the lgsvl simulator docs. What is the output of nvidia-smi if any?

Hi, are you referring to the path here:"~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log"? 
if yes,  it seems not exit so far, I guess maybe the player.log can only be built after the simulator is successfully install.

ivanchristina, 2020-06-19:
@EricBoiseLGSVL  if GPU is mandatory for the simulator, then the virtual machine (Orcale VirtualBox I used)might be the key, will first figure out GPU issue in Orcale VirtualBox, seems somehow impossible after first checking .

only see one run it [https://github.com/lgsvl/simulator/issues/477](url)

closed 2020-06-19 by ivanchristina

kilocharlievictor, 2020-07-10:
Hi,
1)I am trying to run this on a Linux Host(No VM). But i do not have a graphics card. Is graphics card REQUIRED or is it 'NICE TO HAVE' option? . If i am building this with Unity Hub, using the source folder from this page, will that also not work if the Graphics card is not there? I added the Environment and Vehicle asset bundle, but none of them are detected for the build. WebUI built successfully and the Simulator also built, but does not show any assets even though I cloned them in proper directories(vehicle,environment etc)? Please advise.
2)Can this Linux version of the simulator work on ARM64(aarch64) NVIDIA NANO GPU? If graphics card is a must to run this simulator can i run this on Nvidia Nano. It is linux host 64, but aarch64 , Not x86. Please advise.

EricBoiseLGSVL, 2020-07-10:
@kilocharlievictor A GPU is required and it must support Vulkan.  We recommend at least 8gb GPU memory.  Nano is not going to run unity HDRP, sorry.

martins-mozeiko, 2020-07-10:
Also LGSVL Simulator is supposed to be run on desktop computer, not embedded boards. Unity on desktop supports only Intel x86 architecture. Not arm64.

kilocharlievictor, 2020-07-11:
EricBoiseGSVL: Please refer to page 'https://developer.nvidia.com/isaac/downloads'. Can you tell me is this supports HDRP Pipeline through ISAAC on Nano GPU?

martins-mozeiko, 2020-07-12:
@kilocharlievictor Technically it would be possible to get Unity HDRP running on Jetson Nano, but that will require using custom Unity and custom simulator build. By default, public version of Unity does not support running on arm64. Unity has special internal builds that can do that, but you need to talk to Unity to get that, not LGSVL.