lgsvl/simulator #811: About SimCtrl

yumianhuli2, 2020-06-19:
HI！ I have seen the behavior of creating pose based on the vehicle dynamic model. Is there a way to create pose by means of time interpolation called SimCtrl？Thank YOU！

EricBoiseLGSVL, 2020-06-19:
I'm not quite sure what you are asking for here.  Let me add @hadiTab

hadiTab, 2020-06-20:
Hi @yumianhuli2 . Can you provide more details? What do you mean by "create a pose"? Also, I don't know what SimCtrl is.

yumianhuli2, 2020-06-20:
> Hi @yumianhuli2 . Can you provide more details? What do you mean by "create a pose"? Also, I don't know what SimCtrl is.

There are two main ways to create pose: one is based on vehicle dynamics, and the other is based on velocity planning path points. These pathpoints carry timestamps, so new poses can be obtained by interpolation calculation based on the timestamps.

closed 2020-06-20 by yumianhuli2