lgsvl/simulator #812: The traffic signal is red all the time

yumianhuli2, 2020-06-19:
[image]
The traffic signal is always red, so the car stops at the intersection all the time. So How to solve this problem？

ntutangyun, 2020-06-19:
@yumianhuli2 please check out this [issue](https://github.com/lgsvl/simulator/issues/627) and see if you're encountering same problem.

lemketron, 2020-06-19:
The `Traffic Light` module is not enabled in your screen shot, but needs to be in order to enable traffic light detection. Is it crashing or was it just disabled?
[image]

yumianhuli2, 2020-06-22:
> The `Traffic Light` module is not enabled in your screen shot, but needs to be in order to enable traffic light detection. Is it crashing or was it just disabled?
> [image]

Hello！ The traffic Light button automatically jumps to the off state

luqiang21, 2020-06-22:
@yumianhuli2 It is possible that your GPU is not enough to run both `traffic light` and `perception` modules. You can try to run only `traffic light` to see whether it will work without running `perception`.

lemketron, 2020-06-22:
Can you show the output of `nvidia-smi`?

yumianhuli2, 2020-06-22:
> Can you show the output of `nvidia-smi`?
Hello！
[image]

yumianhuli2, 2020-06-22:
> @yumianhuli2 It is possible that your GPU is not enough to run both `traffic light` and `perception` modules. You can try to run only `traffic light` to see whether it will work without running `perception`.

[image]
Traffic Light will still turn off automatically, and vehicle won't move even if I sending the routing request. Could you provide some error messages or logs for it?

lemketron, 2020-06-22:
You appear to only have one "mainboard" instance using the GPU, and I see that the "Perception" and "Traffic Light" switches are both disabled. Watch the "memory usage" number in the middle of `nvidia-smi` screen while you are trying to launch perception and traffic light. I would suggest running `watch -n1 nvidia-smi` so you can watch the GPU memory usage number change as you attempt to enable the modules. 

Enable Perception, which will slowly ramp to just over 2GB and then settle back down, then enable "Traffic Light" and see where the GPU memory ends up -- or see if either module crashes again. 

If a module is crashing you might want to look at the Apollo module logs. If the GPU is running out of memory, then you might try running a built simulator binary rather than running it from Unity. Another option would be to run the simulator on another machine which leaves the entire GPU for use by Apollo.

yumianhuli2, 2020-06-22:
> You appear to only have one "mainboard" instance using the GPU, and I see that the "Perception" and "Traffic Light" switches are both disabled. Watch the "memory usage" number in the middle of [...]

I think what you said is right. So for developers who don't have good machines, Could you develop a Unity simulator that only tests PNC modules?

lemketron, 2020-06-22:
> I think what you said is right. So for developers who don't have good machines, Could you plz Develop a feature, such as only testing the PNC module?

Our apollo instructions are inteded to show how to run the full apollo stack running end to end turning sensor data from the simulator into control data to drive the simulated vehicle.

You can of course enable whichever apollo modules you want, and test them however you want. The simulator generates whatever sensor data you specify in the vehicle configuration. The (e.g. perception, etc.) module(s) that you enable in Apollo process that sensor data and feed it to other modules.

If you are looking for a way to feed perception results to other downstream apollo modules (rather than having to have apollo perception running) that could potentially use the 3D ground truth "sensor" in the simulator, but you'd likely have to implement that yourself unless someone in the apollo community has done it already. You might take a look at the dummy perception and fake prediction modules in apollo as they might be helpful.

yumianhuli2, 2020-06-22:
> Our apollo instructions are inteded to show how to run the full apollo stack running end to end turning sensor data from the simulator into control data to drive the simulated vehicle.
>
> You can of [...]

OK,Thanks

lemketron, 2020-06-22:
> When I played Apollo3.5, I didn't get so stuck. Is the reason for unity HDRP？
> 
> And does the DreamViewer button control whether Apollo module messages are opened or not？

Apollo 5.0 is more demanding on the CPU than 3.5 was. In vehicle configurations (depending on the number of sensors and modules used) it may require a more powerful computer, or an additional computer than what was specified for Apollo 3.5 -- and those are dedicated computers in the vehicle, not computers that are also running Unity and a simulator. That's why the first thing to try is to just run the simulator on a separate (Linux or Windows) computer...

The DreamView module control switches launch the "mainboard" processes that run the various apollo modules, though they can also be started manually from the command line. You can find several [how-to documents](https://github.com/ApolloAuto/apollo/tree/master/docs/howto) in the [apollo github docs](https://github.com/ApolloAuto/apollo/tree/master/docs).

yumianhuli2, 2020-06-22:
> > When I played Apollo3.5, I didn't get so stuck. Is the reason for unity HDRP？
> > And does the DreamViewer button control whether Apollo module messages are opened or not？
> 
> Apollo 5.0 is more demanding on the CPU than 3.5 was. In vehicle configurations (depending on the number of sensors and modules used) it may require a more powerful computer, or even a [second IPC computer](https://github.com/ApolloAuto/apollo/blob/master/docs/howto/how_to_setup_dual_ipc.md) similar to the one specified for Apollo 3.5 -- and those are dedicated computers in the vehicle, not computers that are also running Unity and a simulator. That's why the first thing to try is to just run the simulator on a separate (Linux or Windows) computer...
> 
> The DreamView module control switches launch the "mainboard" processes that run the various apollo modules, though they can also be started manually from the command line. You can find several [how-to documents](https://github.com/ApolloAuto/apollo/tree/master/docs/howto) in the [apollo github docs](https://github.com/ApolloAuto/apollo/tree/master/docs).

57/5000

So in theory, I don't have the vehicle's sensors in webUI and I turn off Apollo's sensing module, so I can have one machine running smoothly.

rongguodong, 2020-06-22:
@yumianhuli2 To better understand why perception module cannot start, you can check its log under /apollo/data/log, or manually start it by using "cyber_launch start /apollo/modules/perception/production/launch/perception.launch" (more details see: https://github.com/ApolloAuto/apollo/blob/master/docs/howto/how_to_run_perception_module_on_your_local_computer.md).

If you manually start it, you should see errors in the console telling you what's wrong with it.

lemketron, 2020-07-25:
> If you are looking for a way to feed perception results to other downstream apollo modules (rather than having to have apollo perception running) that could potentially use the 3D ground truth [...]

Regarding bypassing perception to test only PNC, you might find this thread interesting where code is posted to do just that: https://github.com/lgsvl/simulator/issues/864#issuecomment-661609781
As noted in that same thread, we are also looking at incorporating this functionality into the simulator.

yumianhuli2, 2020-07-31:
> > If you are looking for a way to feed perception results to other downstream apollo modules (rather than having to have apollo perception running) that could potentially use the 3D ground truth "sensor" in the simulator, but you'd likely have to implement that yourself unless someone in the apollo community has done it already. You might take a look at the dummy perception and fake prediction modules in apollo as they might be helpful.
> 
> Regarding bypassing perception to test only PNC, you might find this thread interesting where code is posted to do just that: [#864 (comment)](https://github.com/lgsvl/simulator/issues/864#issuecomment-661609781)
> As noted in that same thread, we are also looking at incorporating this functionality into the simulator.

Very helpful and useful feature!Thank you!

closed 2020-08-28 by EricBoiseLGSVL

lemketron, 2020-10-06:
Note: This new [Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/) feature (which uses ground truth from simulator instead of running Apollo perception and traffic light modules is now built-in as of the 2020.06 release.

If you have not tried this, I would encourage you to take a look since it makes it much easier to run Apollo and LGSVL Simulator on a single machine by publishing ground truth information for perception obstacles as well as traffic light information. This works with Apollo 5.0 and Apollo master.