lgsvl/simulator #814: WebUI build is completed, but not work

AlexandrZabolotny, 2020-06-19:
Hello!
My OS - Ubuntu 18.04

When I follow step 12 `In Unity Editor run Build WebUi... in Unity: Simulator -> Build WebUI...`
I see in comsol

```
WebUI build is completed
UnityEngine.Debug:Log(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:60)
```
**But I can not see "build dialog"**
And next - when I follow step 13 `Open Build... in Unity: Simulator -> Build...` 
**I see error**

Exception: WebUI files are missing! Please build WebUI at least once before building Player
Simulator.Editor.Build.RunPlayerBuild (Simulator.Editor.Build+BuildTarget target, System.String folder, System.Boolean development) (at Assets/Scripts/Editor/Build.cs:737)
Simulator.Editor.Build.OnGUI () (at Assets/Scripts/Editor/Build.cs:563)
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <437ba245d8404784b9fbab9b439ac908>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <437ba245d8404784b9fbab9b439ac908>:0)
System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) (at <437ba245d8404784b9fbab9b439ac908>:0)
UnityEditor.HostView.Invoke (System.String methodName, System.Object obj) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:367)
UnityEditor.HostView.Invoke (System.String methodName) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:361)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:337)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect) (at /home/builduser/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:356)
UnityEditor.DockArea.OldOnGUI () (at /home/builduser/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:346)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize, System.Boolean canAffectFocus) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:293)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect, System.Boolean canAffectFocus) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:503)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, System.Boolean canAffectFocus) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:486)
UnityEngine.UIElements.IMGUIContainer.SendEventToIMGUI (UnityEngine.UIElements.EventBase evt, System.Boolean canAffectFocus) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:477)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:449)
UnityEngine.UIElements.CallbackEventHandler.HandleEventAtTargetPhase (UnityEngine.UIElements.EventBase evt) (at /home/builduser/buildslave/unity/build/Modules/UIElements/Events/EventHandler.cs:77)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/Events/MouseCaptureDispatchingStrategy.cs:92)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:277)
UnityEngine.UIElements.EventDispatcher.ProcessEventQueue () (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:238)
UnityEngine.UIElements.EventDispatcher.OpenGate () (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:203)
UnityEngine.UIElements.EventDispatcherGate.Dispose () (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:44)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:310)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:159)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at /home/builduser/buildslave/unity/build/Modules/UIElements/Panel.cs:256)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/UIElementsUtility.bindings.cs:404)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at /home/builduser/buildslave/unity/build/Modules/UIElements/UIElementsUtility.bindings.cs:194)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at /home/builduser/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:187)

EricBoiseLGSVL, 2020-06-19:
@AlexandrZabolotny Hello and welcome to LGSVL Simulator.  Please see #759 and see if it helps

AlexandrZabolotny, 2020-06-22:
Thanks!
I reinstall npm
1. sudo apt install npm
Next in /simulator/WebUI I open terminal
2. npm install
3. npm pack

And all work

closed 2020-06-22 by EricBoiseLGSVL