lgsvl/simulator #816: Using LGSVL with Autoware : cant't build Docker containers leveraging NVIDIA GPUs

Eolion, 2020-06-22:
Hello, I want to install LGSVL with autoware on my computer under ubuntu. But the command : 
docker run --gpus all nvidia/cuda:10.0-base nvidia-smi doesn't work.

My terminal displays : 
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
ERRO[0026] error waiting for container: context canceled

I try many things even reinstalling ubuntu. But nothing change i always have same error and I can't continue install Autoware.
You can see the result when I use nvdia-smi :
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.132                Driver Version: 390.132                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 770     Off  | 00000000:01:00.0 N/A |                  N/A |
| 34%   34C    P8    N/A /  N/A |    341MiB /  1994MiB |     N/A      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0                    Not Supported                                       |
+-----------------------------------------------------------------------------+

I specify that I am obliged to use the nvidia-390 drivers because the others make display at lgsvl a white screen on which I can do nothing.

Does anyone have the solution to my problem?

dr563105, 2020-06-22:
First, did you install docker properly? Are you able to run other docker commands well? Is the docker version > 19.03?
Then, did you install nvidia docker, properly? -- https://github.com/NVIDIA/nvidia-docker#ubuntu-160418042004-debian-jessiestretchbuster

If you have, then can you try this? -- https://github.com/NVIDIA/nvidia-docker/issues/1034#issuecomment-520282450

Eolion, 2020-06-22:
Yes I follow all steps : "Installation methods" to install docker. 
Command "docker run hello-wold" works.

I try with Driver version Nvidia-440 and terminal don't displays error just the result of nvidia-smi
but after that when I try "Autoware/docker/generic/build.sh kinetic"  my terminal displays this :
 
Use kinetic
./build.sh: 7: ./build.sh: nvidia-docker: not found

Even if it works I don't want to use nvidia-440 because LGSVL doesn't works with this.

dr563105, 2020-06-22:
Okay docker works. 

Leave Nvidia-440 for now. Have the driver version you wish installed. 

These problems are nvidia-docker related. You need to have them properly installed before you go into autoware.

So, 
Did you install nvidia-docker(https://github.com/NVIDIA/nvidia-docker#ubuntu-160418042004-debian-jessiestretchbuster) properly? Have you by chance installed nvidia-docker2?  Which Ubuntu version are you using? In particular did you execute these commands? 
```
sudo apt-get update && sudo apt-get install -y nvidia-container-toolkit
sudo systemctl restart docker
```
After this, can you execute - `docker run --gpus all nvidia/cuda:9.0-base nvidia-smi`
Is the output of that command, exactly like when you execute `nvidia-smi`?

After that, make changes to run.sh as mentioned here --https://www.lgsvlsimulator.com/docs/autoware-instructions/#install-autoware and see if you can run autoware. 

If you run into problems then also, try running the commands mentioned here -- https://github.com/NVIDIA/nvidia-docker/issues/1034#issuecomment-520282450

Then start over with autoware. Despite all these you get an error, you may need to try this -- https://github.com/NVIDIA/nvidia-container-runtime.

Eolion, 2020-06-22:
Thank you ! To fix my problem I just install nvidia-docker2

Now I can execute "sudo build.sh kinetic" but I have another error : 
sometimes red line appears with this message : 

debconf: delaying package configuration, since apt-utils is not installed

W: GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: The repository 'http://packages.ros.org/ros/ubuntu xenial InRelease' is not signed.

And at the end : 

E: There were unauthenticated packages and -y was used without --allow-unauthenticated
The command '/bin/sh -c apt-get update && apt-get install -y ros-kinetic-desktop-full ros-kinetic-nmea-msgs ros-kinetic-nmea-navsat-driver ros-kinetic-sound-play ros-kinetic-jsk-visualization ros-kinetic-grid-map ros-kinetic-gps-common' returned a non-zero code: 100

dr563105, 2020-06-22:
The error you're getting is a common GPG error. You can overcome them by using installation steps given ros install section for the corresponding ros version page.

Eolion, 2020-06-22:
Although I did add the key with the command: 
curl -sSL 'http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xC1CF6E31E6BADE8868B172B4F42ED6FBAB17C654' | sudo apt-key add -

I try : sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
as shown on this link  : https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454  
too

I have exactly same error

dr563105, 2020-06-23:
After adding the key did you do the 'final step' mentioned here -- https://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/ ?

Eolion, 2020-06-23:
I just added --allow-unauthenticated in dockerfile.kinetic after that no problem with key and install.
Thank you for your time and your answers. I close this issue

closed 2020-06-23 by Eolion