lgsvl/simulator #818: Running latest Apollo with LGSVL Simulator

AlexandrZabolotny, 2020-06-23:
Where I can get Vehicle JSON Configuration ( Lincoln2017MKZ)  for running latest Apollo with LGSVL Simulator.

If I will use [https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/](https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/)  

Should I make changes to Apollo settings in this case?

lemketron, 2020-06-23:
> Where I can get Vehicle JSON Configuration ( Lincoln2017MKZ) for running latest Apollo with LGSVL Simulator.
> If I will use https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/
> Should I make changes to Apollo settings in this case?

The default `Lincoln2017MKZ (Apollo 5.0)` vehicle (and the apollo5-0-json-example settings) should work fine for latest Apollo.

AlexandrZabolotny, 2020-06-23:
Have you tried building a HD-map with these settings using /apollo/scripts/msf_simple_map_creator.sh?

rongguodong, 2020-06-23:
No. This is one of the reasons why we cannot use MSF localization but RTK.
According to this page (https://github.com/ApolloAuto/apollo/blob/master/docs/howto/how_to_generate_local_map_for_MSF_localization_module.md) we need to have some dataset to have this script work. But we are not clear what kind of dataset it requires. It is not clearly explained in Apollo's document.

closed 2020-06-25 by AlexandrZabolotny