lgsvl/simulator #819: Create simulation environment based on high-precision maps in Apollo format

ifond, 2020-06-23:
Digital twins of particular environments,
How can I create a map for simulation based on a high-precision map of the real world? Thank you

lemketron, 2020-06-26:
@ifond Can you tell us more about what you are wanting to do?  Are you saying that you have an HD map and you'd like to import that into the simulator? An HD map contains only high-precision information about the roads in the environment, but does not include 3D features needed to render a full "digital twin". There are several ways to create a digital twin but it is not something you can just import.  If you can say more about the information you have available, and what you are trying to do then we can try to help from there. Thanks!

ifond, 2020-06-28:
@lemketron yes,I have an HD map and I'd like to import that into the simulator.I only have high-definition maps here, how do I do it?thank you very much.

lemketron, 2020-06-28:
You can import an HD map into the simulator as described [here](https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations) but that will not create a drivable mesh or 3D scene. The HD map contains only information about the roads and lanes.

If you want to see how to create a new environment, [this video](https://youtu.be/S0w2ahhEPbE) might help.

I'm guessing you'd like to be able to import the HD map and create a basic environment from it. That is a common request and we are looking into that but cannot promise anything at this time.

ifond, 2020-06-28:
@lemketron Yes, your guess is correct,I hope to be able import the HD map and create a basic environment from it.Excuse me, what does "drivable mesh" mean?What is the difference from the lane line of the high-definition map? I think that the vehicle can also plan navigation through the high-definition map.

ifond, 2020-06-28:
@lemketron It can be understood as importing a high-precision map, then creating and editing the simulation environment under this high-precision map framework, how much of this workload is possible, can it be called a reference workload evaluation method, thank you very much

ifond, 2020-06-28:
@lemketron Personally feel that it is very attractive to import high-definition maps and then create a function corresponding to the urban simulation environment

EricBoiseLGSVL, 2020-06-29:
@ifond We are working on this exact feature and plan to release it soon.  Look for our next release.

lycismail, 2020-08-19:
@EricBoiseLGSVL  Excuse me. Have u resolved this issue already?

EricBoiseLGSVL, 2020-08-19:
@lycismail We are working on our next release now and this feature will be included.

MaisJamal, 2020-09-25:
I would like to check, when will be the new release? will it be soon?

EricBoiseLGSVL, 2020-09-25:
We have released the ability to import HDmaps and create a drive-able mesh in Unity with the latest release.  If you are looking for a more detailed Digital Twin, we have services in beta currently that can possibly do this.  @zelenkovsky can let you know who to contact

closed 2020-10-20 by EricBoiseLGSVL