lgsvl/simulator #820: Add an external camera to the Simulator

ashwinsarvesh, 2020-06-23:
Hello all,

I would like to add an external camera to the simulator. How can I do this?
It is an ADAS (Advanced driver assistance systems) camera used for Lane departure warning, forward collision warning, and Traffic sign recognition.

Should I follow the steps in this link? https://www.lgsvlsimulator.com/docs/sensor-plugins/ 
Is ROS Driver of the camera, a mandatory need to use the camera with the Simulator and Apollo?

EricBoiseLGSVL, 2020-06-23:
@ashwinsarvesh Yes, a custom sensor should be a plugin and that link is correct.  You can support any AD stack you want.  If you don't need Apollo support, then no need to add.

rongguodong, 2020-06-23:
Can you say more about the camera? What special feature does it have?

If you simply want to add one normal camera to the ego car, you can add one in JSON configuration. We have ColorCameraSensor, DepthCameraSensor, and SegmentationCameraSensor ready for your use.

rongguodong, 2020-06-23:
If you want to get camera input for perception modules to do lane detection, forward collision warning, and traffic sign recognition, you may simply use one or more ColorCamerSensor for that purpose.

If you want to output ground truth values of those, what you need is not a camera sensor, but some kind of ground truth sensor. We already have a SignalSensor for traffic sign. You can build your own LaneDeparatureSensor plug-in and ForwardCollisionWarningSensor plug-in.

ashwinsarvesh, 2020-06-23:
Thank you so much for instant replies!!

https://www.amazon.de/MOBILEYE-630-PRO-mit-Vibrationsalarm/dp/B01I8M6K28 
This is the camera.

So, my main aim is to somehow use this ADAS camera and its features(mentioned above) on the simulated ego vehicle instead of the actual on-road vehicle. For this, I need to connect this camera to the simulator/apollo. But the problem is there is no ROS driver available for it. So I want to know if this is still possible?

We use the LGSVL simulator with Apollo 3.0 connected via the ROS bridge interface.

ashwinsarvesh, 2020-06-24:
> @ashwinsarvesh Yes, a custom sensor should be a plugin and that link is correct. You can support any AD stack you want. If you don't need Apollo support, then no need to add.

Hi, Do you mean that if I do not need Apollo support, then I do not need to have a ROS driver for the camera?

EricBoiseLGSVL, 2020-06-24:
@ashwinsarvesh no, sorry I didn't know you where running Apollo 3.0.  If you are using ROS then you have the topic to get the data.  I'm not sure how to get Apollo to recognize this data though, @rongguodong ideas?

ashwinsarvesh, 2020-07-15:
> If you want to get camera input for perception modules to do lane detection, forward collision warning, and traffic sign recognition, you may simply use one or more ColorCamerSensor for that [...]

Hello,
Does Apollo 3.0 have the features Lane departure warning, forward collision warning, and Traffic sign recognition?

rongguodong, 2020-07-15:
@ashwinsarvesh You can ask questions about Apollo at [Apollo's GitHub](https://github.com/ApolloAuto/apollo).

ashwinsarvesh, 2020-07-17:
ok thank you