lgsvl/simulator #822: LGSVL crash with some sensors

Eolion, 2020-06-24:
Hello,

I installed LGSVL under ubuntu 18.04 and some simulation by default don't works. I tried many manipulation and I discovered that some sensors making crash the simulator. For example with Lexus2016RXHybrid (Autoware) vehicle : main camera and IMU sensors make my simulator imediatly crash.

My environment:
    NVIDIA GTX770 Geforce
    16 gb RAM
    8 core CPU
    Ubuntu 18.04

I precise on windows I don't have this problem

Here you can see my log file after crash with BorregasAve (with Autoware) simulation  : 
[error.txt](https://github.com/lgsvl/simulator/files/4824410/error.txt)

Does anyone have a solution to this problem?

EricBoiseLGSVL, 2020-06-24:
@Eolion I see that the GPU is way below spec to run simulator.  Windows has better performance with Unity so it makes sense that it crashes on Linux.  We recommend a GPU with 8gb memory.  It could be other driver issues as well, @lemketron any ideas?

yumianhuli2, 2020-06-29:
> @Eolion I see that the GPU is way below spec to run simulator. Windows has better performance with Unity so it makes sense that it crashes on Linux. We recommend a GPU with 8gb memory. It could be other driver issues as well, @lemketron any ideas?

my PC nvidia 8gb is not enough on linux ,12gb may be ok,and 2 pc connect is recommend.

EricBoiseLGSVL, 2020-06-29:
@yumianhuli2 We hope to release new maps that are much less demanding on GPU.  Look for these in the next release

lemketron, 2020-06-29:
> my PC nvidia-GPU 8gb is not enough on linux ,12gb may be ok,and 2 pc connect is recommend.

@yumianhuli2 Can you run `nvidia-smi` and see if there are other applications (e.g. Chrome?) using GPU memory? You should be able to run the simulator with 8GB on Linux, but it is often helpful to run the AD stack (or other software) on a separate system as you mentioned.

yumianhuli2, 2020-07-01:
> @yumianhuli2 Can you run `nvidia-smi` and see if there are other applications (e.g. Chrome?) using GPU memory? You should be able to run the simulator with 8GB on Linux, but it is often helpful to [...]

The other applications shown in Nvidia-SMi take up only a few hundred megabytes.But under the 8G graphics card, the Apollo algorithm module always crashes...

yumianhuli2, 2020-07-01:
> @yumianhuli2 We hope to release new maps that are much less demanding on GPU. Look for these in the next release

This is a great update, and I'm looking forward to running this simulator on an 8GB or even 6GB GPU machine

lemketron, 2020-07-01:
> The other applications shown in Nvidia-SMi take up only a few hundred megabytes.But under the 8G graphics card, the Apollo algorithm module always crashes...

You mentioned having a GTX770... I'm not sure if Apollo supports that. I wonder if Apollo is crashing because of the 770 and not because it's running out of GPU memory. I've been able to run Simulator AND Apollo on a newer 8GB GPU without the Apollo modules crashing.

Put another way, if the problem is Apollo modules crashing, please try testing Apollo without running the simulator to see whether the modules still crash.

lemketron, 2020-07-31:
@yumianhuli2 Any update on this?

closed 2020-08-28 by EricBoiseLGSVL