lgsvl/simulator #823: Sensor Script for Sensor Plug in

ashwinsarvesh, 2020-06-24:
Hello,

In this link https://www.lgsvlsimulator.com/docs/sensor-plugins/ , it says that the sensor script must be added to the root of the prefab. 

1. How to create a sensor script for my Camera which I want to plug into the LGSVL?
2. What camera-specific data do I need to create such a sensor script for my camera?

Kindly help me!

dr563105, 2020-06-24:
Hey
You can look into these repos -- https://github.com/lgsvl/LaneFollowingSensor, https://github.com/lgsvl/ComfortSensor to know exactly how to work with sensor plugins. For camera-specific data, you would need to first figure all the variables and their ROS message types. Then convert them so that LGSVL could understand. For example in LanefollowingSensor, they use TwistStamped message type.

ashwinsarvesh, 2020-06-24:
@dr563105 So, by variables do you mean x,y,z, pitch, yaw, roll and parameter values?

dr563105, 2020-06-24:
Yes. The sensor parameters from JSON, you would want your custom plugin to read, do some computation internally and send it across through ROS. For example, look at this https://github.com/lgsvl/ComfortSensor/blob/master/ComfortSensor.cs . Several parameters are fetched from JSON and some from the vehicle or other inbuilt sensors, do some calculations and then convert ROS bridge understandable format and send. You can also have a look at the built in camera sensor in the simulator code, to understand better.

ashwinsarvesh, 2020-06-24:
@dr563105  Sure, but if I just want to make the camera work with the LGSVL but not need any ROS communications to Apollo, then I can use "Nobridge" and enter "null" for the PUBLISH and SUBSCRIBE topics rite?

ashwinsarvesh, 2020-06-24:
Sorry if the questions are silly, I am really new to all this

dr563105, 2020-06-24:
I haven't worked with Apollo. So, I don't know what you have to pass as arguments. May be someone else can answer you.
However, would you not want to send those images to Apollo? CyberRT is another communication method.

lemketron, 2020-06-24:
@ashwinsarvesh Can you say more about what it is that you are trying to do? Are you using the built-in camera sensors and asking about the JSON sensor parameters? Or are you trying to create your own custom camera sensor in Unity (and if so, why do you feel the need to do that)? And what is the goal for the data generated by the camera sensor (if not to send it over a sensor bridge)?

ashwinsarvesh, 2020-06-29:
@lemketron @dr563105 I am using an ADAS camera that has Lane departure warning, traffic sign recognition, forward collision warning. I should integrate this smart camera to the LGSVL so that the input data from the simulator is fed to the camera and it uses the above features on the incoming data. I do this to evaluate camera performance.

I feel it is enough to add it as a sensor plugin without communication with Apollo. But I want the confirmation from the experts here.

ashwinsarvesh, 2020-06-29:
> I haven't worked with Apollo. So, I don't know what you have to pass as arguments. May be someone else can answer you.
> However, would you not want to send those images to Apollo? CyberRT is another communication method.

Due to the reasons mentioned in my other comment, I do not want to send those images to Apollo.

ashwinsarvesh, 2020-06-29:
> @ashwinsarvesh Can you say more about what it is that you are trying to do? Are you using the built-in camera sensors and asking about the JSON sensor parameters? Or are you trying to create your [...]

I am trying to create my own custom plugin for the camera I have. The reason and goal are mentioned in my other comment.

EricBoiseLGSVL, 2020-10-20:
You can create communication with the camera drivers to your sensor plugin

closed 2020-10-20 by EricBoiseLGSVL