lgsvl/simulator #824: Lanefollowing and LSTM - time series issue in callback part -- drive.py

dr563105, 2020-06-26:
Hey everyone!

I extended the [Lanefollowing](https://github.com/lgsvl/lanefollowing) to use LSTM. Here I split the input images from the 3 camera sensors into a time series of time steps 2. Everything is fine until I use the trained model to predict the commands in the autonomous driving part([drive.py](https://github.com/lgsvl/lanefollowing/blob/master/ros2_ws/src/lane_following/drive.py)). 

I use the ros2 message filters -- ApproximateTimeSynchronizer to approximately sync the 3 images with the control cmds and do callback in that. Just like how it is done in the [collect.py](https://github.com/lgsvl/lanefollowing/blob/master/ros2_ws/src/lane_following/collect.py#L29-L30). 

For time series the model expects 2 times the 3 images that is `2*([left center right])` -- 6 images in total to do prediction. Or making it dynamic with timesteps defined. However at each callback only three images come through. 

So, how can I make the callback wait till the subscriber nodes receive 2 of each images and then do the prediction. Have you tried something like this in ROS? In my limited search, I found that this behaviour is still not implemented  -- 
1. https://answers.ros.org/question/294881/ros2-how-can-i-call-a-service-and-wait-for-the-response-from-within-a-subscriber-callback/ 
2. https://answers.ros.org/question/322831/ros2-wait-for-action-using-rclpy/

Even if I have to use a cruel solution like a boolean flag, how to efficiently implement that? 

Any ideas will be appreciated. Thank you.

EricBoiseLGSVL, 2020-06-26:
@hadiTab any ideas?

dr563105, 2020-06-30:
Solved it. Used a FIFO queue with max length equal to time steps. Appended new images in each callback and invoked prediction only when the queue's length was equal to time steps.

closed 2020-06-30 by dr563105