lgsvl/simulator #825: How can I connect the ros2 in windows

Eolion, 2020-06-26:
Hello,
I need to control a vehicle using the ROS controls. From windows it is not possible for me to launch the docker by following the information from this documentation:  https://www.lgsvlsimulator.com/docs/create-ros2-ad-stack/

How can I connect ros2 bridge to my simulation.For windows ?

dr563105, 2020-06-26:
see issue [754](https://github.com/lgsvl/simulator/issues/754)

Eolion, 2020-06-26:
I have already read this issue. This is just for ros1 and you can't control vehicle by command. I saw it's only possible with ros 2 but the installation documentation provided requires to use dockers-CE. However, the installation of Ubuntu by Microsoft store on Windows does not allow the use of docker-CE.

dr563105, 2020-06-26:
Indeed that guide is for ros1. You need to change where ever it is necessary to ros2. 

Lanefollowing docker image is needed for that tutorial just to establish `rosbridge`. You can do that manually, outside of docker with [ros2webbridge](https://github.com/RobotWebTools/ros2-web-bridge) but I think you might run into issues installing it. If you do install it successfully, you can run the bridge in one terminal and then in another terminal launch the node you want. Moreover, the tutorial is outdated as the latest docker image is with dashing not crystal. You would need to change those environment commands too. 
If rosbridge is established between LGSVL and the host, you can then of course give commands to the vehicle -- publish it in the right topic.