lgsvl/simulator #826: Unable to reach localhost:8080 -- asking for login

dr563105, 2020-06-26:
Hey all,

I cleared data and cache of Mozilla firefox today. Since then, I'm not able to get into the WebUI. When I click 'open browser' in the simulator, I'm asked for a login. Usually I think this gets automatically rerouted to localhost but not happening now. Giving `localhost:8080` gets me to the login landing page. I have already tried deleting the `.config` and restarting the machine. Issue still present. 
Should I have to sign up an account? or is there anyother way? 
[image]

Thanks.

lemketron, 2020-06-26:
Yes, you need to register for an account to use the simulator, but it's fast and easy to do so... thanks!

dr563105, 2020-06-26:
Thanks. I think I registered and forgot to note down the credentials. I tried to use the forgot password feature but I don’t seem to receive any mail. It’s been about two hours.

closed 2020-06-26 by dr563105

lemketron, 2020-06-26:
Often when a "forgot password" email is requested, it gets rejected with an error message such as (in this case):
> Delivery has failed to these recipients or groups:
> dr******@r***-a*****.**
> A problem occurred while delivering this message to this email address. Try sending this message again. If the problem continues, please contact your helpdesk.
> The following organization rejected your message: mx1.**.****-******.**.

dr563105, 2020-06-26:
hah. I understand. I used a different domain now. All good.