lgsvl/simulator #827: Aborted (core dumped) when running on ubnut 18.04

HemaZ, 2020-06-26:
Hello lgsvl, 
when I try to run the simulator on my machine which runs ubuntu 18.04 
CPU: Intel i5-7200U (4) @ 3.100GHz 
GPU: NVIDIA GeForce 920MX 
GPU: Intel Integrated Graphics 
Memory: 11882MiB 
I've the latest nvidia drivers and vulkan
I can open http://localhost:8080 but when I try to run any simulation it shutdowns.

here is my log file 

> `Mono path[0] = '/home/hema/Desktop/lgsvlsimulator-linux64-2020.03-rc1/simulator_Data/Managed'
> Mono config path = '/home/hema/Desktop/lgsvlsimulator-linux64-2020.03-rc1/simulator_Data/MonoBleedingEdge/etc'
> Preloaded 'liblaszip.so'
> Preloaded 'libpng.so'
> Preloaded 'libsqlite3.so'
> Preloaded 'libturbojpeg.so'
> Display 0 '0': 1366x768 (primary device).
> Display 1 'HP 24f 24"': 1920x1080 (secondary device).
> Loading player data from /home/hema/Desktop/lgsvlsimulator-linux64-2020.03-rc1/simulator_Data/data.unity3d
> Desktop is 1366 x 768 @ 60 Hz
> [Vulkan init] extensions: count=18
> [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_display, enabled=1
> [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
> Vulkan detection: 2
> Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
> [Subsystems] Discovering subsystems at path /home/hema/Desktop/lgsvlsimulator-linux64-2020.03-rc1/simulator_Data/UnitySubsystems
> GfxDevice: creating device client; threaded=1
> [Vulkan init] extensions: count=18
> [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_display, enabled=1
> [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
> [Vulkan init] extensions: count=73
> [Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
> [Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
> [Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
> [Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[... 1543 lines cut ...]
> #26 0x00000000413ac6e8 in ?? ()
> #27 0xfffffffffffffffe in ?? ()
> #28 0x00000000ffffd834 in ?? ()
> #29 0x0000000000000000 in ?? ()
> [Inferior 1 (process 14753) detached]
> 
> =================================================================
> Got a SIGSEGV while executing native code. This usually indicates
> a fatal error in the mono runtime or one of the native libraries 
> used by your application.
> =================================================================
> 
> Caught fatal signal - signo:11 code:1 errno:0 addr:0xa0
> Obtained 9 stack frames.
> #0  0x007fb40810c371 in vk::DataBuffer::CreateResource()
> #1  0x007fb40810c037 in vk::DataBuffer::Initialize(void const*, vk::CommandBuffer*, vk::ScratchBuffer*, unsigned long)
> #2  0x007fb4080c3d8d in GfxDeviceVK::InitializeBufferInternal(GfxBuffer*, void const*, GfxUpdateBufferFlags)
> #3  0x007fb408055c9b in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
> #4  0x007fb408056c3b in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
> #5  0x007fb40804c825 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
> #6  0x007fb408447eea in Thread::RunThreadWrapper(void*)
> #7  0x007fb4071ec6db in start_thread
> #8  0x007fb406f1588f in clone
> `
>

martins-mozeiko, 2020-06-27:
Crash on `k::DataBuffer::CreateResource` most likely means running out of GPU memory. 
GeForce 920MX seems to have only 2GB of memory. That's probably not enough for Unity's HDRP rendering for simulation. It may run for very simple scenes (low res textures, simple meshes), but for environment simulator is using more memory is required. Using GPU with at least 8GB is recommended for Simulator.

HemaZ, 2020-06-28:
ok, but can you update the docs and add minimum requirements or something.

EricBoiseLGSVL, 2020-06-29:
@HemaZ  https://www.lgsvlsimulator.com/docs/getting-started/

lemketron, 2020-06-29:
> @HemaZ https://www.lgsvlsimulator.com/docs/getting-started/

More specifically: https://www.lgsvlsimulator.com/docs/getting-started/#getting-started

True these are "recommended" but it's hard to say what the "minimum" requirements are as we do not have access to older generation GPUs, and GPU memory requirements vary based on other applications that may be running as well as the GPU requirements of the desired 3D environment being used in the simulator.

That said, I believe it's best to stick with a 10x0 series, with at LEAST 4GB of GPU memory to run JUST the simulator with a basic environment and nothing else running. You can check GPU memory usage with the `nvidia-smi` tool.

dr563105, 2020-06-29:
I'm having the exact same issue from last night. I have 2 RTX 2080 Ti graphic cards of 11GB each and still the simulator crashes if a python script(ran from docker) also using GPU is already running. Only solution I have right now is to start simulator first and then run the script. 

Also possible because HDD crashed today and had to do recovery. Docker service was affected. Had to reinstall docker. Fortunately, nvidia-docker and runtime container are ok. Using 2020.03 Simulator. 

Can you please help?

Attaching player log. 
[LGSVLplayer.log](https://github.com/lgsvl/simulator/files/4847390/LGSVLplayer.log)

martins-mozeiko, 2020-06-29:
@dr563105 As your log shows same call stack on crash, please monitor GPU memory usage when running simulator. Check that it does not run out of memory. It maybe that you have some other software running that uses too much memory. [nvtop](https://github.com/Syllo/nvtop) is nice utility to use for monitoring memory on Nvidia GPU.

dr563105, 2020-06-29:
Thanks @martins-mozeiko. You were right. Used `nvidia-smi` to monitor the GPU memory. When I ran the script, both GPU memories were full. Now I changed the script to use one and Sim the other. Running fine now.

closed 2020-06-29 by lemketron

marcusvinicius178, 2021-02-07:
Hi I had the same issue. My RAM is below:

lspci -v -s 01:00.0
[Player.log](https://github.com/lgsvl/simulator/files/5938328/Player.log)

01:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
	Subsystem: Acer Incorporated [ALI] GP107M [GeForce GTX 1050 Ti Mobile]
	Flags: bus master, fast devsel, latency 0, IRQ 133
	Memory at a3000000 (32-bit, non-prefetchable) [size=16M]
	Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Memory at a0000000 (64-bit, prefetchable) [size=32M]
	I/O ports at 4000 [size=128]
	Capabilities: <access denied>
	Kernel driver in use: nvidia
lspci: Unable to load libkmod resources: error -12

Also running:
**$nvidia-smi:**
NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 105...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   31C    P8    N/A /  N/A |      0MiB /  4042MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage   

running:
$**lspci** 

lspci00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 05)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller (rev 31)
00:14.2 Signal processing controller: Intel Corporation 100 Series/C230 Series Chipset Family Thermal Subsystem (rev 31)
00:15.0 Signal processing controller: Intel Corporation 100 Series/C230 Series Chipset Family Serial IO I2C Controller #0 (rev 31)
00:15.1 Signal processing controller: Intel Corporation 100 Series/C230 Series Chipset Family Serial IO I2C Controller #1 (rev 31)
00:16.0 Communication controller: Intel Corporation 100 Series/C230 Series Chipset Family MEI Controller #1 (rev 31)
00:17.0 SATA controller: Intel Corporation HM170/QM170 Chipset SATA Controller [AHCI Mode] (rev 31)
00:1c.0 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #3 (rev f1)
00:1c.3 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #4 (rev f1)
00:1f.0 ISA bridge: Intel Corporation HM175 Chipset LPC/eSPI Controller (rev 31)
00:1f.2 Memory controller: Intel Corporation 100 Series/C230 Series Chipset Family Power Management Controller (rev 31)
00:1f.3 Audio device: Intel Corporation CM238 HD Audio Controller (rev 31)
00:1f.4 SMBus: Intel Corporation 100 Series/C230 Series Chipset Family SMBus (rev 31)
01:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
02:00.0 Network controller: Intel Corporation Wireless 7265 (rev 59)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)

**My log file is attached below:**
[Player.log](https://github.com/lgsvl/simulator/files/5938327/Player.log)
I will appreciate if you support me to fix this core(dumped) error when I launch the lgsv simulator

**$Found path: /opt/lgsvl/simulator**
Aborted (core dumped)

It starts opening and then fail unfortunately...

lemketron, 2021-02-07:
> **My log file is attached below:**
> [Player.log](https://github.com/lgsvl/simulator/files/5938327/Player.log)
> I will appreciate if you support me to fix this core(dumped) error when I launch the lgsv simulator

@marcusvinicius178 I'm not sure why the simulator is aborting so early in the launch process. Looks like there may be some issue with Vulkan on your system.

It's possible you might have "mesa-vulkan-drivers" installed. If so, then you can remove that with `sudo apt remove mesa-vulkan-drivers`.

You can also try installing vulkan-tools (`sudo apt install vulkan-tools`). Then type `vulkaninfo` to verify that libvulkan is working, and then see if `vkcube` displays a window with a spinning 3D cube.

marcusvinicius178, 2021-02-07:
Hi Steve thanks for replying I have restarted from scratch and now the
simulator is working 😄.

spicker23, 2021-03-27:
> @marcusvinicius178 I'm not sure why the simulator is aborting so early in the launch process. Looks like there may be some issue with Vulkan on your system.
>
> It's possible you might have [...]

I faced the same problem. Do I need to remove it for ade or for whole system?

EricBoiseLGSVL, 2021-03-29:
The whole system

marcusvinicius178, 2021-03-29:
Hi there are 2 simulators.
One simulador inside ade that the creators of Autoware used to make easy
their life to test demos and packages, etc.this simulator works with rviz2
using ros2 native bridge and there are not ROS- REAL car actuator messages
conversion. To use this imternal simulador inside Ade (the one suitable to
run their demos on website) then you have 2 steps:
1- modify your .aderc file and remove the line that says something about
"Nvidia drivers = True" remove this line and source .aderc file before
enter the image.
2- sudo apt-get remove mesa-vulkan-drivers

The issue here is that ALWAYS when you exit the Virtual machine (Ade image)
you will need to repeat the second command above and remove the drivers
again. Because when you start the image the drivers are there again and you
need to remove.

And also a simulator outside ade ( that you must download from Lg website).
This external simulator is on your host system and works with rviz2 using
ros2 bridge on vehicles configuration.
For the simulador outside ade there are just one step:
1- sudo apt-get remove mesa-vulkan-drivers

Later is just configure the ros2-lgsvl-bridge to use with rviz2 and modify
the rviz2 topics on vehicles urdf in LG simulator or set them properly in
rviz2 based on LG vehicles urdf because the tutorials of Autoware Auto
writes them for the internal simulator, that has different topics for
lidar, sensors, etc

I hope it works.