lgsvl/simulator #828: Error response from daemon

yumianhuli2, 2020-06-28:
I changed a PC to built Apollo5.0, but the following error occurred:
XXXXXXXXXXXX:./docker/scripts/dev_into.sh 
Error response from daemon: Container 39c2b25e12c9195f5a3fdd0ca151cebb7f60212ffa70d1dc4d8d5058d7fb80d8 is not running
[image]

[image]

lemketron, 2020-06-28:
According to our [Building Apollo](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#building-apollo-and-bridge) instructions, you need to first launch the container before you can enter it, by typing:
`./docker/scripts/dev_start.sh`

yumianhuli2, 2020-06-28:
> According to the [Building Apollo](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#building-apollo-and-bridge) instructions, you need to first launch the container before you can enter it, by typing:
> `./docker/scripts/dev_start.sh`

Hello！I have already typed:
./docker/scripts/dev_start.sh

lemketron, 2020-06-28:
Please try typing: `./docker/scripts/dev_start.sh stop`
Then type: `./docker/scripts/dev_start.sh`

If you can share the console output from both of these that might help...

yumianhuli2, 2020-06-28:
> Please try typing: `./docker/scripts/dev_start.sh stop`
> Then type: `./docker/scripts/dev_start.sh`
> 
> If you can share the console output from both of these that might help...

Ｈello!
First step  :
 ./docker/scripts/dev_start.sh
[sudo] password for XXX: 
[INFO] Start docker container based on local image : lgsvl/apollo-5.0:latest
yolo3d_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:33add717a675be26c86f817d2207c1ab1ec25694f0decb63e55f0fc3d14b1397
Status: Image is up to date for apolloauto/apollo:yolo3d_volume-x86_64-latest
docker.io/apolloauto/apollo:yolo3d_volume-x86_64-latest
374e936134d3b7b3c45f6099bdbadbe5773545e74b2d504cfa2f799ac763f348
localization_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:d8be22d8221c784e8a29af14a23be7fc5f2164137ec24228701e6fdcb689b088
Status: Image is up to date for apolloauto/apollo:localization_volume-x86_64-latest
docker.io/apolloauto/apollo:localization_volume-x86_64-latest
e8015590186f0582076be999c4559edd7568fd53df22ecff4fa0215792961e4a
paddlepaddle_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:901f110a4f13501e074f599b2c7d636a7584d290f59ef50237a7d77e66d025fb
Status: Image is up to date for apolloauto/apollo:paddlepaddle_volume-x86_64-latest
docker.io/apolloauto/apollo:paddlepaddle_volume-x86_64-latest
03d3a883e000499149d2171593d8f29289b6d3ce436b1e7942cd396e7b2ff5d7
local_third_party_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:b844c40ab0af0733ce69c38b220a8b14f752bd834f63343392ce8508b98b546b
Status: Image is up to date for apolloauto/apollo:local_third_party_volume-x86_64-latest
docker.io/apolloauto/apollo:local_third_party_volume-x86_64-latest
2c9ca92bd25f4856607b761f63f17a93dc563701310d5c9cc922096dd35bcdfb
[INFO] Starting docker container "apollo_dev_XXX" ...
./docker/scripts/dev_start.sh: line 311: type: nvidia-docker: not found
++ local_volumes
++ volumes='-v /home/XXX/Documents/apollo-5.0:/apollo              -v /home/XXX/.cache:/home/XXX/.cache'
++ case "$(uname -s)" in
+++ uname -s
++ volumes='-v /home/XXX/Documents/apollo-5.0:/apollo              -v /home/XXX/.cache:/home/XXX/.cache -v /dev:/dev                                 -v /media:/media                                 -v /tmp/.X11-unix:/tmp/.X11-unix:rw                                 -v /etc/localtime:/etc/localtime:ro                                 -v /usr/src:/usr/src                                 -v /lib/modules:/lib/modules'
++ echo '-v /home/XXX/Documents/apollo-5.0:/apollo              -v /home/XXX/.cache:/home/XXX/.cache -v /dev:/dev                                 -v /media:/media                                 -v /tmp/.X11-unix:/tmp/.X11-unix:rw                                 -v /etc/localtime:/etc/localtime:ro                                 -v /usr/src:/usr/src                                 -v /lib/modules:/lib/modules'
+ docker run -it -d --privileged --gpus all --name apollo_dev_XXX --volumes-from apollo_yolo3d_volume_XXX--volumes-from apollo_localization_volume_XXX--volumes-from apollo_paddlepaddle_volume_XXX--volumes-from apollo_local_third_party_volume_XXX -e DISPLAY=:0 -e DOCKER_USER=XXX -e USER=XXX -e DOCKER_USER_ID=1000 -e DOCKER_GRP=apollo -e DOCKER_GRP_ID=1000 -e DOCKER_IMG=lgsvl/apollo-5.0:latest -e USE_GPU=1 -v /home/XXX/Documents/apollo-5.0:/apollo -v /home/XXX.cache:/home/XXX/.cache -v /dev:/dev -v /media:/media -v /tmp/.X11-unix:/tmp/.X11-unix:rw -v /etc/localtime:/etc/localtime:ro -v /usr/src:/usr/src -v /lib/modules:/lib/modules --net host -w /apollo --add-host in_dev_docker:127.0.0.1 --add-host XXX-NH5x-7xEDx-RCx-RDx:127.0.0.1 --hostname in_dev_docker --shm-size 2G --pid=host -v /dev/null:/dev/raw1394 lgsvl/apollo-5.0:latest /bin/bash
789417fff290b57348ebe27b98fcb296cb8adc26f7f6a3464fef5412b353fc8c
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
+ set +x
Error response from daemon: Container 789417fff290b57348ebe27b98fcb296cb8adc26f7f6a3464fef5412b353fc8c is not running
[ OK ] Finished setting up Apollo docker environment. Now you can enter with: 
bash docker/scripts/dev_into.sh
[ OK ] Enjoy!

Second step:
 ./docker/scripts/dev_into.sh
Error response from daemon: Container 789417fff290b57348ebe27b98fcb296cb8adc26f7f6a3464fef5412b353fc8c is not running

Third step:
./docker/scripts/dev_start.sh stop
stopping container: apollo_local_third_party_volume_XXX ...       [DONE]
stopping container: apollo_paddlepaddle_volume_XXX ...            [DONE]
stopping container: apollo_localization_volume_XXX ...            [DONE]
stopping container: apollo_yolo3d_volume_XXX ...                  [DONE]

Fourth step:
./docker/scripts/dev_start.sh
[INFO] Start docker container based on local image : lgsvl/apollo-5.0:latest
yolo3d_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:33add717a675be26c86f817d2207c1ab1ec25694f0decb63e55f0fc3d14b1397
Status: Image is up to date for apolloauto/apollo:yolo3d_volume-x86_64-latest
docker.io/apolloauto/apollo:yolo3d_volume-x86_64-latest
b34438f0b7d53e7e7b3e94445d97a6833d5ce011e4cb6da36a609205fcce64df
localization_volume-x86_64-latest: Pulling from apolloauto/apollo
Digest: sha256:d8be22d8221c784e8a29af14a23be7fc5f2164137ec24228701e6fdcb689b088
Status: Image is up to date for apolloauto/apollo:localization_volume-x86_64-latest
docker.io/apolloauto/apollo:localization_volume-x86_64-latest
[... 6 lines cut ...]
local_third_party_volume-x86_64-latest: Pulling from apolloauto/apollo
[... cut ...]
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
+ set +x
Error response from daemon: Container 2031b472c8f534c74bf9bb22b98b5c89ed3cb505057b83edd53fe27fb18563a9 is not running
[ OK ] Finished setting up Apollo docker environment. Now you can enter with: 
bash docker/scripts/dev_into.sh
[ OK ] Enjoy!

Fifth step:
./docker/scripts/dev_into.sh
Error response from daemon: Container 2031b472c8f534c74bf9bb22b98b5c89ed3cb505057b83edd53fe27fb18563a9 is not running

lemketron, 2020-06-28:
This appears to be the problem:
`docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].`

Did you install Docker CE and nvidia-docker as described [here](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#docker)?

Can you also try running `nvidia-smi`?

And try `nvidia-docker run nvidia/cuda:10.0-base nvidia-smi` which should help verify if nvidia-docker is installed.

yumianhuli2, 2020-06-28:
> This appears to be the problem:
> `docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].`
>
> Did you install Docker CE and nvidia-docker as described [...]

Hello！
Plz see this：
First:
In ubuntu18.04，Kernel version 5.3 is too high and Nvidia-SMI has problems.

So I dropped the kernel version to 4.x, and running Nvidia-SMi was OK.

Second:
I tried nvidia-docker run nvidia/cuda:10.0-base nvidia-smi
I got nvidia-docker:command not found
But:
[image]

martins-mozeiko, 2020-06-28:
You don't have nvidia docker installed. That is an error when you run `scripts/docker/dev_start.sh`

See here in your output:
[image]

Make sure you install nvidia-docker. Refer to its documentation for installation instructions for ubuntu version you use.

yumianhuli2, 2020-06-28:
> You don't have nvidia docker installed. That is an error when you run `scripts/docker/dev_start.sh`
> 
> See here in your output:
> [image]
> 
> Make sure you install [nvidia-docker](https://github.com/NVIDIA/nvidia-docker). Refer to its documentation for installation instructions for ubuntu version you use.
> 
> Here are instructions from Apollo installation guide: https://github.com/ApolloAuto/apollo/blob/master/docs/quickstart/apollo_software_installation_guide.md#Set-up-the-Docker-environment

Got it!

yumianhuli2, 2020-06-28:
> This appears to be the problem:
> `docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].`
>
> Did you install Docker CE and nvidia-docker as described [...]

Got it!

closed 2020-06-28 by lemketron