lgsvl/simulator #829: why is the image from simulator so dark

qwetqwe, 2020-06-28:
my sensor parameter is as follows.I use apollo cyber_visualizer to see the image from simulator camera.But the image is so dark. What is the reason? I did not see where to configure brightness in sensor parameter
 {
      "type": "Color Camera",
      "name": "Main Camera",
      "params": {
         "Width": 1920,
         "Height": 1080,
         "Frequency": 20,
         "JpegQuality": 100,
         "FieldOfView": 50,
         "MinDistance": 0.1,
         "MaxDistance": 1000,
         "Topic": "/apollo/sensor/camera/front_6mm/image/compressed",
         "Frame": "front_6mm"
      },
      "parent": "Lidar",
      "transform": {
         "x": 0,
         "y": 1.7,
         "z": -0.2,
         "pitch": 0,
         "yaw": -90,
         "roll": 0
      }
   },
{
      "type": "Color Camera",
      "name": "Telephoto Camera",
      "params": {
        "Width": 1920,
        "Height": 1080,
        "Frequency": 20,
        "JpegQuality": 100,
        "FieldOfView": 10,
        "MinDistance": 0.1,
        "MaxDistance": 1000,
        "Topic": "/apollo/sensor/camera/front_12mm/image/compressed"
      },
      "transform": {
        "x": 0,
        "y": 1.7,
        "z": -0.2,
        "pitch": -4,
        "yaw": 0,
        "roll": 0
      }
    }
[image]

rongguodong, 2020-06-28:
Is the left part of the screenshot from cyber_visualizer? cyber_visualizer does show the color of compressed image correctly. You can try the uncompressed channels to see the correct color.

qwetqwe, 2020-06-29:
@rongguodong 
Thanks for your reply. RGB image channel is same as that, dark too. Is there any other way to configure brightness?

qsengshu, 2020-06-29:
Anyone found a solution to this problem?

Lanedetection is currently not possible because the image is too dark.

Best regards,
qsengshu

rongguodong, 2020-06-29:
Such kind of darkness is usually because of sRGB<->linear color space mismatch. Not sure if it is a bug in our code or a bug in cyber_visualizer. We will investigate it soon.

rongguodong, 2020-06-29:
@qsengshu It seems a regression in your latest version (2020.05). If you do not need new features in 2020.05, you can temporarily use 2020.03, which should not have this dark issue.

dr563105, 2020-07-06:
For my project I need the changes done in 2020.05. I can of course use 2020.03 but I need to do the changes manually. Could it be possible to give us a workaround, please, if the new release is sometime away? Also the car's hood looks weird when the camera image is taken, processed and evaluated. Can you check that as well, please?

[image]

rongguodong, 2020-07-06:
A temporary workaround is to use a distorted camera with all parameters set to 0:
{
    "type": "Color Camera",
    "name": "Main Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 1000,
      "Topic": "/simulator/main_camera",
      "Frame": "camera",
      "Distorted": true,
      "DistortionParameters": [
        0, 0, 0, 0
      ]
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}
This should work with 2020.05.

dr563105, 2020-07-06:
Thank you. It is little bright but not bright enough for the ego to drive without misjudging the lanes. Anyother change that could make it work? Can you also give me a fix so that the car's hood looks red than purple-ish, please? I'd like to record the simulation.

rongguodong, 2020-07-06:
You can try to change [this line](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/SensorRenderTarget.cs#L104) yourself: changing the format to R8G8B8A8_UNorm to R8G8B8A8_SRGB.

dr563105, 2020-07-08:
@rongguodong I tried it and there is only a slight improvement. The front hood is still over exposed and lighting on the player, changes drastically upon pressing 'play'; shadows and open area change the look of the ego as well.

EricBoiseLGSVL, 2020-07-08:
@dr563105 We hope to make a release with these changes soon.  For now you could attempt to adjust the post processing effects to reduce exposure and shorten exposure adjustment time.

closed 2020-08-28 by EricBoiseLGSVL