lgsvl/simulator #830: CarSim and Physic Material

BestMediaDeveloper, 2020-06-30:
My company has now succeeded in getting Mecanical Simulation's CarSim to run in Unity, and we've published a script for it.
https://github.com/BestMediaDeveloper/VsDotnet

We have successfully incorporated it into the LGSV simulator.
https://github.com/BestMediaDeveloper/VsDotnet/blob/master/Documents/LGSVL.md

I have one question . How do I change the friction coefficient of the PhysicMaterial for the road?
The coefficient of friction is currently low at 0.6 and I'd like to change it to 0.8 for asphalt, but I didn't know that. 

We have another request.

In IVehicleDynamics.cs, there is a Rigidbody. It seems to be used to get speed and acceleration and angular acceleration, but the RigidBody is It's Unity specific. to use different dynamics to get the speed etc. directly instead of via Rigidbody. I want to.

EricBoiseLGSVL, 2020-06-30:
@BestMediaDeveloper Excellent.  This is great that you integrated CarSim. Excellent work.
- You can create a Physics Material and apply it to the road Mesh Collider component. See [here](https://docs.unity3d.com/Manual/class-PhysicMaterial.html)
- We provide VehicleSMI as a starting point for dynamics.  You can create a dynamics class that uses IVehicleDynamics but toggle the rigidbody to kinematic and write code to move the transform.  Then you will need to calculate the speed etc manually.  We do this for some of our sensors.  If you look at ExampleVehicleFMU, there is an option for Non-Unity solver as a reference but you wouldn't need to inherit VehicleFMU.

BestMediaDeveloper, 2020-06-30:
@EricBoiseLGSVL  

Thank you. just set it up the usual way. I thought it was difficult to do it with programs and so on.

Yes, I'm currently modifying IVehicleDynamics a bit and creating a new CarSimSMI It is created and used.
Speed and other data can be obtained directly from CarSim, so it is outputted.

The changes are
https://github.com/BestMediaDeveloper/VsDotnet/tree/master/Samples/LGSVL

But this is just as easy to do here.

closed 2020-06-30 by BestMediaDeveloper