lgsvl/simulator #833: OSM to apollo opendrive: resampling issue

nju044, 2020-07-02:
When we use map conversion feature to convert lanelet2 map to apollo opendrive format, we find the point number descrease to about 20% of the original lanelet2 map. Can anyone offer a solution to maintain the number of point while converting lanelet2 to apollo opendrive?
BTW, does LGSVL open the map conversion source code to users like me ? I don't know C#, so it's hard for me to locate where the map conversion source code is. 

Thanks in advance!

luqiang21, 2020-07-02:
@nju044 Our importers and exporters are implemented to provide a way to interface our internal data structure with other map formats. We can't guarantee the conversion to be very accurate due to information missing when importing from and exporting into different formats. 

For importing, you can set both thresholds on the Map Importer window to be 0 to keep all points.  But Unity may crash to show too many lane's points. The source code for map importers and exporters are located in [here](https://github.com/lgsvl/simulator/tree/master/Assets/Scripts/Editor). For example, lanelet2 importer is [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Lanelet2MapImporter.cs).

nju044, 2020-07-08:
> @nju044 Our importers and exporters are implemented to provide a way to interface our internal data structure with other map formats. We can't guarantee the conversion to be very accurate due to [...]

@luqiang21 Thanks for your info. We will wander around the codes.

closed 2020-07-08 by nju044