lgsvl/simulator #834: Need suggestions for camera integration

ashwinsarvesh, 2020-07-02:
Hello all,
I need to use a real camera instead of the available virtual cameras in the simulator like ColorCameraSensor, DepthCameraSensor, and SegmentationCameraSensor.
I should place this camera in front of the simulation screen. It should capture the images and send it to apollo.(just like how it is sent now from the virtual cameras). Then apollo needs to process this data to do lane detection, forward collision warning, and traffic sign recognition.
Can you suggest some real cameras, which I can integrate in the loop to Apollo for the above purpose? 

Main goal of my project is to evaluate the performance of the actual camera by creating different simulation scenarios.

ashwinsarvesh, 2020-07-02:
I am using LGSVL simulator which is interfaced by ROS bridge to Apollo 3.0

rongguodong, 2020-07-02:
It sounds nothing to do with simulator: you are using a physical camera, and you want to send its data to Apollo. You can refer to Apollo document to see how Apollo works with real camera. If you still have problems, you can raise an issue to Apollo.