lgsvl/simulator #835: Interpreting output from depth camera

Doodle1106, 2020-07-03:
Hi,

I came into some questions that I hope you guys can help to clarify:

1. what is the meaning of each pixel of the output of depth camera? corresponding disparity or depth?
2. what is the intrinstic parameters(Cx, Cy, Fx, Fy) of the main camera? where can I modify them?
3. is it possible to add a  pair of stereo cameras with exact the same timestamps? is there a plugin for it?

Thanks for trying to help.
Joey

rongguodong, 2020-07-04:
First of all, concept in LGSVL Simulator are most from computer graphics context, instead of computer vision context. Second, the "depth" camera may be more accurately called "distance" camera, since it output "distance" (i.e. Euclidean distance between a 3D point to camera origin), not "depth" (or "z-value" in computer graphics). So the answers of your questions are:

1. Each pixel in the result records the Euclidean distance between the 3D point corresponding to that pixel and the camera origin. If you want different value, you can simply modify code [here](https://github.com/lgsvl/simulator/blob/master/Assets/Shaders/DepthPass.hlsl#L43).

2. We use perfect pin-hole camera model. So Cx = Width/2, Cy = Height/2, Fx = Fy = (Height/2)/tan(FOV/2) (note FOV in Unity is the vertical FOV, so we use Height here).

3. Yes, you can add two cameras with the distance equal to IPD, and make their frequencies the same. The timestamps of sensors are all from [SimulatorManager.Instance.CurrentTime](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/CameraSensorBase.cs#L384). So if they are published in the same frame, their timestamp will be the same.

Doodle1106, 2020-07-06:
Thanks @rongguodong.

closed 2020-07-06 by Doodle1106