lgsvl/simulator #836: Camera data to the Perception module of Apollo 3.0

ashwinsarvesh, 2020-07-03:
Hello,

How does the camera data from the LGSVL is sent as input to the perception module of Apollo 3.0? What is the mechanism? I could not find it in the documentation.

lemketron, 2020-07-03:
The camera data (for Apollo 3.0) goes over the ROS bridge (not CyberRT as in Apollo 3.5 and later).  The topic that perception is looking at is likely "/apollo/sensor/camera/obstacle/front_6mm" as mentioned in [this document](https://github.com/ApolloAuto/apollo/blob/master/docs/howto/how_to_run_apollo_2.5_perception_visualizer.md) (which is for 2.5 but there may be similar documents for 3.0). By default the simulator will send this information to Apollo perception but as it's not exactly clear what you are trying to do, you'll have to figure out how to send the data from your camera to Apollo either into the perception module or to the downstream module.

I would recommend studying more about apollo and autonomous driving stacks, perhaps through Udacity's self-driving car courses as well as reading all of the documentation available about apollo. Then you can understand what each module does, how they interconnect, and how you can implement whatever you are wanting to do.

The LGSVL Simulator is not an AD stack. It is a proxy for (simulation of) an actual autonomous vehicle. The goal is to render (simulate) sensor data from a real vehicle and present it to a computer that would be able to drive that car. The simulator could also be used to render simulated data that is read by a real-world sensor like a camera, and fed to the self-driving computer, but that is not an actual use case for us at this time. If you want to bypass perception or use a hardware sensor that apollo does not already know how to work with, you'll need to talk with Apollo folks to see if they can help understand how best to do this.

ashwinsarvesh, 2020-07-04:
Thank you. The Apollo guys replied this below:

**Processes**
The whole process is like below:

hardware -> driver -> topic(network) -> perception module
      ↑                   ↑
    camera              lgsvl

1. Lgsvl capture the picture then sent the picture from network by publish the topic to perception module.
2. Camera capture the picture by camera dirvers, and then publish the topic to perception module

ashwinsarvesh, 2020-07-04:
[image]

Thank you
The Apollo guys replied as above