lgsvl/simulator #837: ROS2 missing topics

tgif73, 2020-07-05:
Hello,
I can't seem to subscribe to some of ROS2 messages. LG simulator is publishing lgsvl_msgs but I don't see those lgsvl_msgs although I've compiled those messages successfully. I'm using 2020.05 version with ROS2 dashing.
Is there something else I should do to show those messages?

Thanks for the help.

[image]

rongguodong, 2020-07-06:
Do you have corresponding sensors added to the ego? For example, if you want to publish BoundinggBox2D, you need to add GroundTruth2DSensor.

tgif73, 2020-07-06:
hm, I used default Ego with sample ROS2 json configuration (https://www.lgsvlsimulator.com/docs/autoware-auto-json-example/). The simulator UI shows sensors are publishing...though I don't see those topics on my subscribe node.

EricBoiseLGSVL, 2020-07-07:
@tgif73 GroundTruth2DSensor is not in that list and is not required for working with Autoware.Auto.  If you need the sensor, you can just add to the parameter list.

tgif73, 2020-07-07:
@EricBoiseLGSVL I'm more interested in subscribing to lgsvl/state_report topic (for CanBusData) at the moment. The issue I'm having is, as you can see in the bottom right terminal window, not all published topics from lgsvl simulator is shown on my node...

EricBoiseLGSVL, 2020-07-08:
@hadiTab Do you know why all topics are not publishing?

hadiTab, 2020-07-08:
@tgif73 after building lgsvl_msgs did you run `source <path_to_lgsvl_msgs>/install/setup.bash` in the terminal running the ros2 web bridge?

tgif73, 2020-07-08:
@hadiTab  I didn't previously but I tried again with sourcing both ros2 and lgsvl msg before starting the ros2 bridge. It still has the same issue... I don't see all topics...I see lgsvl_msgs with 'ros2 msg list' command
[image]

hadiTab, 2020-07-08:
@tgif73 Please launch the ros2 web bridge using: `DEBUG=ros2-web-bridge* node bin/rosbridge.js 2> log.txt` 

Let it run for a few seconds (while the simulator is running) and then stop it using Ctrl+C. You can search the generated log file for errors or just share it here and I will have a look at it.

tgif73, 2020-07-08:
@hadiTab Thanks. I collected the log and see those lgsvl_msgs are not recognized/found. I'm not sure why but all those messages are listed with "ros2 msg list" command - build was successful. 
-------------------------------------------------------------------------------------------
2020-07-08T19:57:40.534Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/lgsvl/state_report","type":"lgsvl_msgs/CanBusData"}
2020-07-08T19:57:40.534Z ros2-web-bridge:Bridge advertise a topic: /lgsvl/state_report
2020-07-08T19:57:40.535Z ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"advertise: Error: The message required does not exist: lgsvl_msgs, msg, CanBusData"}
2020-07-08T19:57:40.549Z ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","topic":"/lgsvl/vehicle_control_cmd","type":"lgsvl_msgs/VehicleControlData"}
2020-07-08T19:57:40.549Z ros2-web-bridge:Bridge subscribe a topic named /lgsvl/vehicle_control_cmd
2020-07-08T19:57:40.550Z ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"subscribe: Error: The message required does not exist: lgsvl_msgs, msg, VehicleControlData"}
2020-07-08T19:57:40.551Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/lidar_rear/points_raw","type":"sensor_msgs/PointCloud2"}
2020-07-08T19:57:40.551Z ros2-web-bridge:Bridge advertise a topic: /lidar_rear/points_raw
2020-07-08T19:57:40.606Z ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /lidar_rear/points_raw.
2020-07-08T19:57:40.606Z ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"}
2020-07-08T19:57:40.606Z ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","topic":"/lgsvl/vehicle_state_cmd","type":"lgsvl_msgs/VehicleStateData"}
2020-07-08T19:57:40.607Z ros2-web-bridge:Bridge subscribe a topic named /lgsvl/vehicle_state_cmd
2020-07-08T19:57:40.607Z ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"subscribe: Error: The message required does not exist: lgsvl_msgs, msg, VehicleStateData"}
2020-07-08T19:57:40.607Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/lidar_front/points_raw","type":"sensor_msgs/PointCloud2"}
2020-07-08T19:57:40.607Z ros2-web-bridge:Bridge advertise a topic: /lidar_front/points_raw
2020-07-08T19:57:40.612Z ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /lidar_front/points_raw.
2020-07-08T19:57:40.612Z ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"}
2020-07-08T19:57:40.612Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/imu/imu_raw","type":"sensor_msgs/Imu"}
2020-07-08T19:57:40.613Z ros2-web-bridge:Bridge advertise a topic: /imu/imu_raw
2020-07-08T19:57:40.638Z ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /imu/imu_raw.
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"}
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/lgsvl/vehicle_odom","type":"autoware_auto_msgs/VehicleOdometry"}
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge advertise a topic: /lgsvl/vehicle_odom
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"advertise: Error: The message required does not exist: autoware_auto_msgs, msg, VehicleOdometry"}
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/lgsvl/gnss_odom","type":"nav_msgs/Odometry"}
2020-07-08T19:57:40.638Z ros2-web-bridge:Bridge advertise a topic: /lgsvl/gnss_odom
2020-07-08T19:57:40.681Z ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /lgsvl/gnss_odom.
2020-07-08T19:57:40.681Z ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"}
2020-07-08T19:57:40.681Z ros2-web-bridge:Bridge JSON command received: {"op":"advertise","topic":"/gnss/fix","type":"sensor_msgs/NavSatFix"}
2020-07-08T19:57:40.681Z ros2-web-bridge:Bridge advertise a topic: /gnss/fix
2020-07-08T19:57:40.714Z ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /gnss/fix.
2020-07-08T19:57:40.714Z ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"}
2020-07-08T19:57:43.299Z ros2-web-bridge:Bridge JSON command received:

hadiTab, 2020-07-08:
@tgif73 I believe you will need to rebuild your ros2-web-bridge and source ros2 and lgsvl_msgs in the terminal before building:
```
cd <path_to_bridge>/ros2-web-bridge
rm -rf .node_modules
source /opt/ros/dashing/setup.bash
source <path_to_lgsvl_msgs>/install/setup.bash
npm install
```

tgif73, 2020-07-08:
@hadiTab Thank you so much. rebuilding ros2-web-bridge following your instruction fixed the issue. I see all topics now. Much appreciated!

closed 2020-07-08 by tgif73